riscv-collab/riscv-gnu-toolchain #531: configure: error: *** The available kernel headers are older than the requested

ssmolov, 2019-10-21:
When compiling the toolchain on my Lubuntu x64 host, the following error appears:
```
cd build-glibc-linux-rv64imafdc-lp64d && \
        CC="riscv64-unknown-linux-gnu-gcc " \
        CXX="riscv64-unknown-linux-gnu-g++ " \
        CFLAGS=" -mcmodel=medlow -g -O2 " \
        CXXFLAGS=" -mcmodel=medlow -g -O2 " \
        ASFLAGS=" -mcmodel=medlow " \
        /home/ssedai/tools/riscv-gnu-toolchain/riscv-glibc/configure \
        --host=riscv64-unknown-linux-gnu \
        --prefix=/usr \
        --disable-werror \
        --enable-shared \
        --enable-obsolete-rpc \
        --with-headers=/home/ssedai/tools/riscv-gnu-toolchain/linux-headers/include \
        --disable-multilib \
        --enable-kernel=3.0.0 \
         \
        --libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking for riscv64-unknown-linux-gnu-gcc... riscv64-unknown-linux-gnu-gcc
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv64-unknown-linux-gnu-gcc  accepts -g... yes
checking for gcc... gcc
checking for riscv64-unknown-linux-gnu-readelf... riscv64-unknown-linux-gnu-readelf
checking whether we are using the GNU C++ compiler... no
checking whether riscv64-unknown-linux-gnu-g++  accepts -g... no
checking whether riscv64-unknown-linux-gnu-g++  can link programs... no
checking for sysdeps preconfigure fragments... aarch64 alpha arm csky hppa i386 m68k microblaze mips nios2 powerpc riscv s390 sh sparc x86_64
checking for use of fpu sysdeps directories... yes
checking for -fstack-protector... yes
checking for -fstack-protector-strong... yes
checking for -fstack-protector-all... yes
checking for assembler and linker STT_GNU_IFUNC support... no
checking for gcc attribute ifunc support... no
checking whether the linker supports textrels along with ifunc... no
checking if compiler warns about alias for function with incompatible types... yes
configure: WARNING: gcc emits a warning for alias between functions of incompatible types
configure: WARNING: Multi-arch is disabled.
checking sysdep dirs... sysdeps/unix/sysv/linux/riscv/rv64 sysdeps/unix/sysv/linux/riscv sysdeps/riscv/nptl sysdeps/unix/sysv/linux/generic sysdeps/unix/sysv/linux/wordsize-64 sysdeps/unix/sysv/linux sysdeps/nptl sysdeps/pthread sysdeps/gnu sysdeps/unix/inet sysdeps/unix/sysv sysdeps/unix sysdeps/posix sysdeps/riscv/rv64/rvd sysdeps/riscv/rv64/rvf sysdeps/riscv/rvf sysdeps/riscv/rvd sysdeps/riscv/rv64 sysdeps/riscv sysdeps/init_array sysdeps/ieee754/ldbl-128 sysdeps/ieee754/dbl-64 sysdeps/ieee754/flt-32 sysdeps/wordsize-64 sysdeps/ieee754 sysdeps/generic
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln -s works... yes
checking whether /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/as is GNU as... yes
checking whether /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld is GNU ld... yes
checking for /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/as... /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/as
checking version of /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/as... 2.32, ok
checking for /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld... /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld
checking version of /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld... 2.32, ok
checking for gnumake... no
checking for gmake... no
checking for make... make
checking version of make... 4.2.1, ok
checking for gnumsgfmt... no
checking for gmsgfmt... no
checking for msgfmt... msgfmt
checking version of msgfmt... 0.19.8.1, ok
checking for makeinfo... makeinfo
checking version of makeinfo... 6.5, ok
checking for sed... /bin/sed
checking version of /bin/sed... 4.5, ok
checking for gawk... /usr/bin/gawk
checking version of /usr/bin/gawk... 4.1.4, ok
checking for bison... bison
checking version of bison... 3.0.4, ok
checking if riscv64-unknown-linux-gnu-gcc  is sufficient to build libc... yes
checking for riscv64-unknown-linux-gnu-nm... riscv64-unknown-linux-gnu-nm
checking for python3... python3
checking version of python3... 3.6.8, ok
checking LD_LIBRARY_PATH variable... ok
checking for bash... /bin/bash
checking for perl... /usr/bin/perl
checking for install-info... /usr/bin/install-info
checking for .set assembler directive... yes
checking linker support for protected data symbol... yes
checking linker support for INSERT in linker script... yes
checking for broken __attribute__((alias()))... no
checking whether to put _rtld_local into .sdata section... yes
checking whether to use .ctors/.dtors header and trailer... no
checking for libunwind-support in compiler... no
checking whether --noexecstack is desirable for .S files... yes
checking for -z combreloc... yes
checking for linker that supports -z execstack... yes
checking for linker that supports --no-dynamic-linker... yes
checking for -static-pie... yes
checking for -fpie... yes
checking for --hash-style option... yes
checking for sufficient default -shared layout... no
checking for GLOB_DAT reloc... no
checking linker output format... elf64-littleriscv
checking for -fno-toplevel-reorder -fno-section-anchors... yes
checking for -mtls-dialect=gnu2... no
checking whether cc puts quotes around section names... no
checking for __builtin_memset... no
checking for redirection of built-in functions... yes
checking for compiler option to disable generation of FMA instructions... -ffp-contract=off
checking if riscv64-unknown-linux-gnu-gcc  accepts -fno-tree-loop-distribute-patterns with __attribute__ ((__optimize__))... yes
checking for libgd... no
checking for is_selinux_enabled in -lselinux... no
checking for _FORTIFY_SOURCE predefine... no
checking whether the linker provides working __ehdr_start... yes
checking for __builtin_trap with no external dependencies... yes
checking whether the C++ compiler supports thread_local... no
running configure fragment for sysdeps/unix/sysv/linux/riscv
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
running configure fragment for sysdeps/unix/sysv/linux
checking installed Linux kernel header files... 3.2.0 or later
configure: WARNING: minimum kernel version reset to 4.15.0
checking for kernel header at least 4.15.0... too old!
configure: error: *** The available kernel headers are older than the requested
*** compatible kernel version
make: *** [Makefile:252: stamps/build-glibc-linux-rv64imafdc-lp64d] Error 1 
```
What can go wrong here?

Thanks in advance.

jim-wilson, 2019-10-24:
What does "git status" in the source tree say?

I'm guessing that you did git clone, git submodule update, git checkout.  I was able to reproduce your problem this way.  That will leave you with an inconsistent tree, because git checkout does not update submodules.  You need to do a git submodule update --init --recursive after the git checkout to get a consistent tree.  If this is the problem, then git status will show (new commits) for binutils, gcc, and other submodules.

ssmolov, 2019-10-25:
It helps, thank you!

closed 2019-10-25 by ssmolov