riscv-collab/riscv-gnu-toolchain #533: gcc toolchain install problems

weedmank, 2019-10-23:
Tried using the following on CentOS7
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
make linux

Got an error about python being to old.  Updated to latest but it still complained. I noticed it had checked for python3 just before and it wasn't installed so I installed the latest python3 and now I'm stuck here:
...
checking if riscv32-unknown-linux-gnu-gcc  is sufficient to build libc... yes
checking for riscv32-unknown-linux-gnu-nm... riscv32-unknown-linux-gnu-nm
checking for python3... python3
checking version of python3... 3.6.8, ok
configure: error: 
*** These critical programs are missing or too old: make
*** Check the INSTALL file for required versions.
make: *** [stamps/build-glibc-linux-rv32gc-ilp32d] Error 1
checking if riscv32-unknown-linux-gnu-gcc  is sufficient to build libc... yes
checking for riscv32-unknown-linux-gnu-nm... riscv32-unknown-linux-gnu-nm
checking for python3... python3
checking version of python3... 3.6.8, ok
configure: error: 
*** These critical programs are missing or too old: make
*** Check the INSTALL file for required versions.
make: *** [stamps/build-glibc-linux-rv32gc-ilp32d] Error 1
[root@localhost riscv-gnu-toolchain]# yum install make
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: mirror.tocici.com
 * epel: mirror.sjc02.svwh.net
 * extras: repo1.sea.innoscale.net
 * updates: centos.sonn.com
Package 1:make-3.82-24.el7.x86_64 already installed and latest version
Nothing to do
[root@localhost riscv-gnu-toolchain]#

aswaterman, 2019-10-23:
Looking at the glibc configure.ac, it appears that project requires make version 4 or later.

weedmank, 2019-10-23:
What's the next step? centOS7 thinks it has the latest and the make says 3.6.8 is OK

weedmank, 2019-10-23:
oh. you said MAKE, ,not python.

weedmank, 2019-10-23:
[root@localhost riscv-gnu-toolchain]# yum update make
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: mirror.tocici.com
 * epel: mirror.sjc02.svwh.net
 * extras: repo1.sea.innoscale.net
 * updates: centos.sonn.com
No packages marked for update

What should I do?

jim-wilson, 2019-10-24:
You can download GNU make sources and build them to use for the toolchain build.  You can check for alternate make packages with different names, e.g. maybe there is a make4 package available.  But if you're OS is so old that you don't have a new enough make, then you may also run into other problems.

make v4 is from 2013 so it isn't that new.  It is Centos 7 that is old.  glibc tends to be aggressive with build tools dependencies, and CentOS tends to be very conservative with tools, so this isn't a very good match.  RISC-V is very new, so there isn't any older glibc version that is useful to use.  If you can use a different machine with a newer OS version that might be a simpler solution.

weedmank, 2019-10-24:
I tried installing make 4.2, but it still acted the same.  I have "CentOS Linux release 7.7.1908 (Core)"

wget https://buildlogs.centos.org/centos/7/sclo/x86_64/rh/devtoolset-7/devtoolset-7-make-4.2.1-2.el7.x86_64.rpm
wget http://mirror.centos.org/centos/7/sclo/x86_64/rh/devtoolset-7/devtoolset-7-runtime-7.1-4.el7.x86_64.rpm
sudo yum localinstall –i devtoolset-7-runtime-7.1-4.el7.x86_64.rpm
sudo yum localinstall –i devtoolset-7-make-4.2.1-2.el7.x86_64.rpm

a0u, 2019-10-24:
@weedmank: Did you enable devtoolset in your environment?

```
$ scl enable devtoolset-7 bash
```

It's also better to install [devtoolset](https://www.softwarecollections.org/en/scls/rhscl/devtoolset-7/) through yum repos rather than local RPMs.  Briefly:
```
$ sudo yum install centos-release-scl
$ sudo yum install devtoolset-7-make
```

weedmank, 2019-10-24:
no I didn't.

weedmank, 2019-10-24:
My linux knowledge is limited...

jim-wilson, 2019-10-24:
My Centos knowledge is limited, but Albert Ou gave a link that explains how to use the devtoolset.  Try following the instructions there.

weedmank, 2019-10-24:
I now have a newer make and it is doing the build. Will update you again when it stops.

[root@localhost riscv-gnu-toolchain]# make --version
GNU Make 4.2.1
Built for x86_64-redhat-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

weedmank, 2019-10-24:
Thank you aOu!  I was able to build the 32-bit RV32GC toolchain!  This issue is solved for me. Thanks to all.

closed 2019-10-24 by weedmank

chenjiuhai, 2023-05-15:
you'd better to find the make command location.  ex: which make
then use the newer make to replace the /bin/make
ex: mv /bin/make /bin/make.old.bak && ln -s **/which/newer/make** /bin/make

TommyMurphyTM1234, 2023-05-15:
> you'd better to find the make command location. ex: which make then use the newer make to replace the /bin/make ex: mv /bin/make /bin/make.old.bak && ln -s **/which/newer/make** /bin/make
> 
> if you don't have the newer make, you can install it via miniconda, ex: conda install -c conda-forge make==4.3. [https://docs.conda.io/en/latest/miniconda.html](url)

Link doesn't work but, in any case, you're replying to a 3+ year old issue today has already been addressed and closed since...