riscv-collab/riscv-gnu-toolchain #535: get an error when running "make linux"

GQBBBB, 2019-10-30:
I get an error when I run `make linux`:
```
...
checking for gnumake... no
checking for gmake... gmake
checking version of gmake... 3.82, bad
checking for gnumsgfmt... no
checking for gmsgfmt... no
checking for msgfmt... msgfmt
checking version of msgfmt... 0.19.8.1, ok
checking for makeinfo... makeinfo
checking version of makeinfo... 5.1, ok
checking for sed... /bin/sed
checking version of /bin/sed... 4.2.2, ok
checking for gawk... /bin/gawk
checking version of /bin/gawk... 4.0.2, ok
checking for bison... bison
checking version of bison... 3.0.4, ok
checking if riscv64-unknown-linux-gnu-gcc  is sufficient to build libc... yes
checking for riscv64-unknown-linux-gnu-nm... riscv64-unknown-linux-gnu-nm
checking for python3... python3
checking version of python3... 3.6.8, ok
configure: error: 
*** These critical programs are missing or too old: make
*** Check the INSTALL file for required versions.
make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 1
```

I download a newer gmake, install it to a location like /opt/make, But it doesn't work.
```bash
cd ~
curl -O http://ftp.gnu.org/gnu/make/make-3.82.tar.gz
tar xvf make-3.82.tar.gz
cd make-3.82
./configure --prefix=/opt/make && make && make install
export PATH=/opt/make/bin:$PATh
ln -s /opt/make/bin/make /opt/make/bin/gmake
```

Can you help me?
Thank you!!!

GQBBBB, 2019-10-31:
hello, I installed the Developer Toolset as you said, it seems to work, thanks!
```
$ make linux
...
make[12]: Nothing to be done for 'install-data-am'.
make[12]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb/build-gnulib/import'
make[11]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb/build-gnulib/import'
make[10]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb/build-gnulib/import'
make[9]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb/build-gnulib/import'
make[8]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb/build-gnulib'
make[7]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb/build-gnulib'
make[6]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb/build-gnulib'
make[5]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb'
make[4]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb'
make[3]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux/gdb'
make[2]: Nothing to be done for 'install-target'.
make[2]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux'
make[1]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gdb-linux'
mkdir -p stamps/ && touch stamps/build-gdb-linux
```

But I seem to have a new problem, When I use a "hello world" applet to test, it tells me I can't find the header file:
```
$ riscv64-unknown-linux-gnu-gcc hello.c
hello.c:1:10: fatal error: stdio.h: No such file or directory
    1 | #include <stdio.h>
      |          ^~~~~~~~~
compilation terminated.
```

```
$ cat hello.c 
#include <stdio.h>

int main(){
    printf("hello world!\n");
}
```

I found your answer in another issue, after adding `-v`, the result is as follows, it seems that the header file could not be found.
```
[zzf@hw24 ~]$ riscv64-unknown-linux-gnu-gcc hello.c -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/riscv64-unknown-linux-gnu/9.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/zzf/source-code/riscv-gnu-toolchain/build/../riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/usr/local --with-sysroot=/usr/local/sysroot --with-newlib --without-headers 
--disable-shared --disable-threads --with-system-zlib --enable-tls --enable-languages=c --disable-libatomic --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-bootstrap --src=../../riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2  -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2  -mcmodel=medlow'Thread model: single
gcc version 9.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d'
 /usr/local/libexec/gcc/riscv64-unknown-linux-gnu/9.2.0/cc1 -quiet -v hello.c -quiet -dumpbase hello.c -mtune=rocket -march=rv64imafdc -mabi=lp64d -auxbase hello -version -o /tmp/ccsh8Kt1.s
GNU C17 (GCC) version 9.2.0 (riscv64-unknown-linux-gnu)
	compiled by GNU C version 4.8.5 20150623 (Red Hat 4.8.5-39), GMP version 6.0.0, MPFR version 3.1.1, MPC version 1.0.1, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring nonexistent directory "/usr/local/sysroot/usr/local/include"
ignoring nonexistent directory "/usr/local/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/include
 /usr/local/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/include-fixed
 /usr/local/sysroot/usr/include
End of search list.
GNU C17 (GCC) version 9.2.0 (riscv64-unknown-linux-gnu)
	compiled by GNU C version 4.8.5 20150623 (Red Hat 4.8.5-39), GMP version 6.0.0, MPFR version 3.1.1, MPC version 1.0.1, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: 97d2fe7b7f25053fa71e48bf51bdce37
hello.c:1:10: fatal error: stdio.h: No such file or directory
    1 | #include <stdio.h>
      |          ^~~~~~~~~
compilation terminated.
```

jim-wilson, 2019-10-31:
stdio.h should be in /usr/local/sysroot/usr/include.  Did you look in there?  Are there any header files in there?

Are you sure that the build completed successfully?  If you you had a broken build tree and tried to rebuild it is possible that something did not get rebuilt correctly.  Maybe the stage1 gcc built, but glibc did not.  In that case you would have a gcc but not any libraries or header files.  You could try a build from scratch to see if that works.

GQBBBB, 2019-11-01:
There are no header files under `/usr/local/sysroot/usr/include`
```
$ ll /usr/local/sysroot/usr/include
total 52
drwxrwxr-x  2 zzf zzf  4096 Sep 26 17:41 asm
drwxrwxr-x  2 zzf zzf  4096 Sep 26 17:41 asm-generic
drwxrwxr-x  2 zzf zzf  4096 Sep 26 17:41 drm
drwxrwxr-x 28 zzf zzf 16384 Sep 26 17:41 linux
drwxrwxr-x  2 zzf zzf    70 Sep 26 17:41 misc
drwxrwxr-x  2 zzf zzf   135 Sep 26 17:41 mtd
drwxrwxr-x  3 zzf zzf  4096 Sep 26 17:41 rdma
drwxrwxr-x  3 zzf zzf   167 Sep 26 17:41 scsi
drwxrwxr-x  2 zzf zzf  4096 Sep 26 17:41 sound
drwxrwxr-x  2 zzf zzf    89 Sep 26 17:41 video
drwxrwxr-x  2 zzf zzf   110 Sep 26 17:41 xen
```

jim-wilson, 2019-11-01:
Those dirs are all from linux-headers.  Something went wrong with the glibc build.  You need to rebuild it.  You can try "rm stamps/build-glibc-*" followed by make.  But if glibc is broken there could be other stuff broken too.  You probably should rebuild from scratch now that you have the devtools installed.

GQBBBB, 2019-11-01:
I rebuilt it according to the steps in the README file. When I executed `make linux`, it didn't have any error feedback.

When I executed `riscv64-unknown-linux-gnu-gcc hello.c`, it still told me that the header file could not be found. The header file does not exist in the `/usr/local/sysroot/usr/include` directory.

When I run `rm stamps/build-glibc-* & make linux`, the result is as follows, there should be no error,
```
...
----------------------------------------------------------------------
Libraries have been installed in:
   /opt/riscv/riscv64-unknown-linux-gnu/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the `-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the `LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the `LD_RUN_PATH' environment variable
     during linking
   - use the `-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to `/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
 /usr/bin/mkdir -p '/opt/riscv/share/info'
 /usr/bin/install -c -m 644 ./libgomp.info '/opt/riscv/share/info'
 install-info --info-dir='/opt/riscv/share/info' '/opt/riscv/share/info/libgomp.info'
:
:
:
 /usr/bin/mkdir -p '/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/finclude'
 /usr/bin/install -c -m 644 omp_lib.h omp_lib.f90 omp_lib.mod omp_lib_kinds.mod ../../.././riscv-gcc/libgomp/openacc_lib.h ../../.././riscv-gcc/libgomp/openacc.f90 openacc.mod openacc_kinds.mod '/opt/riscv/lib
/gcc/riscv64-unknown-linux-gnu/9.2.0/finclude' /usr/bin/mkdir -p '/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/include'
 /usr/bin/install -c -m 644 omp.h ../../.././riscv-gcc/libgomp/openacc.h '/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/include'
make[5]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgomp'
make[4]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgomp'
make[3]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgomp'
make[3]: Entering directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic'
Makefile:867: warning: overriding recipe for target 'all-multi'
Makefile:861: warning: ignoring old recipe for target 'all-multi'
Making install in testsuite
make[4]: Entering directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic/testsuite'
make[5]: Entering directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic/testsuite'
make[5]: Nothing to be done for 'install-exec-am'.
make[5]: Nothing to be done for 'install-data-am'.
make[5]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic/testsuite'
make[4]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic/testsuite'
make[4]: Entering directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic'
Makefile:867: warning: overriding recipe for target 'all-multi'
Makefile:861: warning: ignoring old recipe for target 'all-multi'
true  DO=all multi-do # make
make[5]: Entering directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic'
Makefile:867: warning: overriding recipe for target 'all-multi'
Makefile:861: warning: ignoring old recipe for target 'all-multi'
true  DO=install multi-do # make
 /usr/bin/mkdir -p '/opt/riscv/riscv64-unknown-linux-gnu/lib'
 /bin/sh ./libtool   --mode=install /usr/bin/install -c   libatomic.la '/opt/riscv/riscv64-unknown-linux-gnu/lib'
libtool: install: /usr/bin/install -c .libs/libatomic.so.1.2.0 /opt/riscv/riscv64-unknown-linux-gnu/lib/libatomic.so.1.2.0
libtool: install: (cd /opt/riscv/riscv64-unknown-linux-gnu/lib && { ln -s -f libatomic.so.1.2.0 libatomic.so.1 || { rm -f libatomic.so.1 && ln -s libatomic.so.1.2.0 libatomic.so.1; }; })
libtool: install: (cd /opt/riscv/riscv64-unknown-linux-gnu/lib && { ln -s -f libatomic.so.1.2.0 libatomic.so || { rm -f libatomic.so && ln -s libatomic.so.1.2.0 libatomic.so; }; })
libtool: install: /usr/bin/install -c .libs/libatomic.lai /opt/riscv/riscv64-unknown-linux-gnu/lib/libatomic.la
libtool: install: /usr/bin/install -c .libs/libatomic.a /opt/riscv/riscv64-unknown-linux-gnu/lib/libatomic.a
libtool: install: chmod 644 /opt/riscv/riscv64-unknown-linux-gnu/lib/libatomic.a
libtool: install: /opt/riscv/riscv64-unknown-linux-gnu/bin/ranlib /opt/riscv/riscv64-unknown-linux-gnu/lib/libatomic.a
libtool: finish: PATH="/home/zzf/source-code/riscv-gnu-toolchain/scripts/wrapper/sed:/opt/riscv/bin:/opt/rh/devtoolset-7/root/usr/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/usr/local/go/bin:/usr/lo
cal/maven3/bin:/home/zzf/.local/bin:/home/zzf/bin:/sbin" ldconfig -n /opt/riscv/riscv64-unknown-linux-gnu/libldconfig: /opt/riscv/riscv64-unknown-linux-gnu/lib/libgcc_s.so.1 is for unknown machine 243.

[... cut ...]
make[3]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libatomic'
make[2]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2'
make[1]: Leaving directory '/home/zzf/source-code/riscv-gnu-toolchain/build-gcc-linux-stage2'
cp -a /opt/riscv/riscv64-unknown-linux-gnu/lib* /opt/riscv/sysroot
mkdir -p stamps/ && touch stamps/build-gcc-linux-stage2
```

But when running `riscv64-unknown-linux-gnu-gcc hello.c`, it still can't find the header file.

jim-wilson, 2019-11-01:
The glibc build output says that the toolchain was installed into /opt/riscv, so we are looking in the wrong place.  But /opt/riscv wasn't in the search list for gcc header files, and /usr/local/ did seem to have the linux kernel headers for the toolchain installed there.  I think there is something wrong on your end and I don't have time to figure it out.  Maybe you accidentally built two riscv64-unknown-linux-gcc compilers with different prefixes, and the one you are running is broken, but the other one is OK?

Maybe it would be easier if you downloaded a pre-built toolchain from somewhere.  If you run a very recent OS distro, like fedora, you can just install a package to get a riscv64-linux cross compiler.

If you want to solve your problem, I think you need to start with rm -rf to remove what you have, and then start over from scratch with the devtools compiler, and see if it works.  And note your configure options in the bug report if you file another bug report.  And you should also rm -rf your riscv toolchain install trees in case there is something broken in there that is causing the build to fail.  if you can't rm -rf your install trees, then just use a different --prefix when configuring that you haven't used before.

GQBBBB, 2019-11-04:
You are right，I accidentally built two riscv64-unknown-linux-gcc compilers in the /usr/local/bin and /opt/riscv/bin directories respectively.

Thanks !!!

closed 2019-11-04 by GQBBBB

StephenFann, 2020-04-27:
make[3]: *** [../o-iterator.mk:9: /Users/mac/riscv64-linux/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/dlfcn/stamp.oS] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/Users/mac/riscv64-linux/riscv-gnu-toolchain/riscv-glibc/dlfcn'
make[2]: *** [Makefile:258: dlfcn/subdir_lib] Error 2
make[2]: Leaving directory '/Users/mac/riscv64-linux/riscv-gnu-toolchain/riscv-glibc'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/Users/mac/riscv64-linux/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d'
make: *** [Makefile:243: stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2

I have this error when I running "make Linux" in the macos

aswaterman, 2020-04-27:
You need a case-insensitive filesystem to build glibc. By default, Mac OS uses a case-sensitive filesystem.

donno2048, 2021-02-18:
@aswaterman 
 > You need a case-sensitive filesystem to build glibc. By default, Mac OS uses a case-insensitive filesystem.

I'm having the same problem on wsl, could that be the reason?

donno2048, 2021-02-18:
For those who may come across this in the future, I manage to solve this by running:
```ps1
fsutil.exe file setCaseSensitiveInfo /Users enable
```
In Administrator PowerShell