riscv-collab/riscv-gnu-toolchain #538: Thread support

fwsGonzo, 2019-11-02:
Hey, I am using latest master to compile test programs for my own 32-bit RISC-V emulator. So far so good. I am having trouble enabling thread support, though? How would you do it here? I tried editing the Makefile after configure to remove all disable-thread options, but no dice. I've tried all the cross-compilers, although I really only expected `riscv32-unknown-linux-gnu-gcc` to work.

Also, will it be possible to use threads with the normal newlib cross-compiler? With some tinkering perhaps?

https://github.com/riscv/riscv-musl/blob/staging/src/thread/riscv64/clone.s
Musl has some pthread code, seems very straight-forward. I just need one combination to enable pthread, preferrably one of the embedded cross-compilers, as the linux riscv-compiler never stops creating RELRO entries which crashes my emulator. :)

aswaterman, 2019-11-02:
Pthread support in glibc (i.e. linux-gnu) is known to work.

fwsGonzo, 2019-11-02:
Yes, unfortunately glibc seems to add RELRO entries.. so I guess I will just have to bite the bullet and implement that during ELF loading.

I have also tried to build musl (which I'm very familiar with), but I don't really know how to build only that?

jim-wilson, 2019-11-02:
There is no RISC-V thread support in newlib yet.  Thread support usually requires some kind of OS support, and no one has defined a standard way for RISC-V embedded elf systems to support threads yet.

"make musl" should build a toolchain using musl, but I don't test this support.  If you edit the makefile to change the all: line to all:musl this should happen by default.  We have a configure option to choose between linux and newlib, but not for musl.

fwsGonzo, 2019-11-16:
Thanks for all the help so far. I have managed to create, jump to a thread and then exit it. However, when the glibc code joins the thread, I get a protection fault:
```
[00014FE4] 00069317 AUIPC TMP1, PC+430080 (0x7DFE4)
[00014FE8] ED432303 LOADW TMP1, [TMP1-300 = 0x7DEB8]
[00014FEC]     9302 C.JALR RA, TMP1
```
This looks an awful lot like a two-step .rela-like function call.. and it's in the `.got`. I haven't been able to find any information on what a RISC-V `.got` is supposed to contain, and its 0xCC in size for me. Readelf gives me no information, and I only have a `.rela.dyn` with one zero-entry.

```
  [18] .got              PROGBITS        0007ddf0 06cdf0 0000cc 04  WA  0   0  4
```
How do I tell which symbol / function is at `.got + 0xC8`?

jim-wilson, 2019-11-16:
You can find ABI info at
  github.com/riscv/riscv-elf-psabi-doc

Call stubs should be in the plt section, and objdump knows how to disassemble them, at least on riscv-linux.  I suppose you could try something like "objdump -dr --full-contents --section=.got <file>" but that may or may not be helpful.  The got should contain what it does on other systems, addresses of global symbols.

I don't know anything about C++ <thread>.

Since the got normally contains data, the segment permissions are only read/write, with no execute permissions.  If you have memory protection, and are handling the segment permissions correctly, and did try to execute code in a data section, then I would expect that to fail.

Maybe you have a problem with the linker script?  Plts should end up in their own section, separate from the text section, but if you have a bad linker script maybe they could accidentally end up someplace else like the got section.

ilg-ul, 2019-11-17:
> could be that the C++ standard library has some dynamic code in it?

the C++ standard library <thread> implementation is basically a wrapper over pthreads.

if by 'dynamic' you mean created at run-time in RAM and executed from there, then the answer is no (as far as I know, given that I wrote a <thread> implementation for µOS++).

if by 'dynamic' you mean dynamically allocated, then yes, the C++ standard library has plenty of dynamic memory allocations.

fwsGonzo, 2019-12-07:
It turned out to be a faulty instruction in my emulator. There was nothing wrong with the toolchain or any of the libraries. The only thing that I had problems with was using rv-sim and rv-jit with complex programs, as they tended to just crash or print the wrong information. I had to use gdb with target sim instead, but that was enough.

fwsGonzo, 2020-02-21:
Alright, this took me way too long to figure out, but here it is. As I wrote in one of my previous posts when you try to delete a C++ thread it jumps to a zero-GOT entry and faults. The reason?

Turns out that if you never reference pthread_join in your code it won't get pulled into the executable, and you can confirm this with grepping for it in the symbol table with any build options. If you pthread_join a standard pthread then it will get referenced, and voila, now C++ threads work fully.

Easy enough to replicate, I suppose.

jim-wilson, 2020-02-21:
You haven't given us any testcase.

glibc pthread support is known to work.

Is this a musl bug?  If so, you should probably post to some musl related group, as I don't think that there are any musl maintainers reading this.

fwsGonzo, 2020-02-21:
Yes, no problems with glibc. This is either a linker issue or a C++ library issue, or a combination. I don't know, but the problem is apparent with readelf.

So, here I have a C pthread_join function call in the code:
```
$ riscv32-unknown-elf-readelf -a hello_world | grep join
   179: 00000000     0 FILE    LOCAL  DEFAULT  ABS pthread_join.c
   180: 00000000     0 FILE    LOCAL  DEFAULT  ABS pthread_join_common.c
  1658: 000122a0    52 FUNC    WEAK   DEFAULT    3 _ZNKSt6thread8joinableEv
  2250: 000145e6     6 FUNC    GLOBAL DEFAULT    3 __pthread_join
  2320: 00019e2a    44 FUNC    GLOBAL DEFAULT    3 _ZNSt6thread4joinEv
  3129: 00014608   288 FUNC    GLOBAL DEFAULT    3 __pthread_timedjoin_ex
  3367: 000145e6     6 FUNC    WEAK   DEFAULT    3 pthread_join
```
I also have a C++ thread that I also join. It works.

After commenting out `pthread_join(t, NULL)`, building again:
```
$ riscv32-unknown-elf-readelf -a hello_world | grep join
  1654: 00012288    52 FUNC    WEAK   DEFAULT    3 _ZNKSt6thread8joinableEv
  2315: 00019c12    44 FUNC    GLOBAL DEFAULT    3 _ZNSt6thread4joinEv
```
How do you join with the C++ thread now? Predictably the program now jumps to a garbage address.

I will post a tiny test program at some point, bear with me! I don't know if this is a problem with RISC-V toolchain.

fwsGonzo, 2020-03-01:
Alright, so I updated to the latest version of this repo and now the problems are gone. I also didn't load the program headers to GLIBC, and maybe that had something to do with it too. :thinking: 

Anyway, all is well. Have a nice weekend.

closed 2020-03-01 by fwsGonzo

fwsGonzo, 2020-03-01:
Unfortunately, I am still having the problem on my website. I solved it for now by adding `-Wl,--undefined pthread_join` to my linker arguments. I don't really know why it's not pulling in that one specific function. Never seen that before.

fwsGonzo, 2022-02-14:
> Unfortunately, I am still having the problem on my website. I solved it for now by adding `-Wl,--undefined pthread_join` to my linker arguments. I don't really know why it's not pulling in that one specific function. Never seen that before.

This one bit me again, this time with the distro provided riscv64-linux-gnu-g++-10. Simply using `thread::join` will cause a jump to 0x0 due to `pthread_join` not being referenced in. The fix is as above: `-pthread -Wl,--undefined pthread_join` .

jim-wilson, 2022-03-07:
I can reproduce your problem by adding -static to the compile.  What is interesting is that I can also reproduce with a native x86_64 linux g++.  So this isn't a RISC-V specific problem.  Looks like a libstdc++ and static library issue.  I do see that the libstdc++ thread::join function has a reloc pointing at pthread_join, so this should work.  But I see that pthread_join is a weak symbol in libc.a so maybe that has something to do with the problem.  I would suggest asking on an FSF GCC mailing list like gcc-help.