riscv-collab/riscv-gnu-toolchain #540: memset in the generated binary

nitish2112, 2019-11-06:
Hi,

I am trying to compile some piece of code which has global arrays. I have initialized all the global arrays to make sure the compiler does not generate the code to initialize them to 0. However, I am still getting these "memset" calls in the _start section of the assembly. This "memset" is as I understand is trying to set some piece of memory to 0. This is seriously slowing down the execution of the program. Is there a way to get rid of this "memset"?

Here is the disassembly:
```
00000000000100e8 <_start>: 
100e8:       00013197                auipc   gp,0x13 
100ec:       8a818193                addi    gp,gp,-1880 # 22990 <__global_pointer$>
100f0:       84818513                addi    a0,gp,-1976 # 221d8 <_edata>
100f4:       43018613                addi    a2,gp,1072 # 22dc0 <_end>
100f8:       40a60633                sub     a2,a2,a0
100fc:       00000593                li      a1,0
10100:       5a4000ef                jal     ra,106a4 <memset>
10104:       00000517                auipc   a0,0x0
10108:       47c50513                addi    a0,a0,1148 # 10580 <__libc_fini_array>
1010c:       00000097                auipc   ra,0x0
10110:       424080e7                jalr    1060(ra) # 10530 <atexit>
10114:       00000097                auipc   ra,0x0
10118:       4e8080e7                jalr    1256(ra) # 105fc <__libc_init_array>
1011c:       00012503                lw      a0,0(sp)
10120:       00810593                addi    a1,sp,8
10124:       00000613                li      a2,0
10128:       100000ef                jal     ra,10228 <main>
1012c:       00000317                auipc   t1,0x0
10130:       41c30067                jr      1052(t1) # 10548 <exit> 
```

Also if my understanding is right, memset is trying to set the memory locations from 0x221d8 to 0x22dc0 (marked as _edata and _end) in the disassembly. These are the addresses which are not being used at all. The global arrays that I have declared are all mapped to addresses less than 0x221d8

This is what the address space looks like:

```
00000000000221d0 <__malloc_trim_threshold>:
   221d0:       0000                    unimp
   221d2:       0002                    0x2
   221d4:       0000                    unimp
        ...

Disassembly of section .sbss:

00000000000221d8 <_PathLocale>:
        ...

00000000000221e0 <__malloc_max_total_mem>:
        ...
```

As it can be seen, there is nothing at 0x221d8. Can someone explain what is going on? And how can I get rid of this memset?

I tried -fno-builtin compiler flag but it didn't help.

Thanks,
Nitish

jim-wilson, 2019-11-06:
This is code in crt0.o to clear bss.  This is a region of memory to be cleared at program start, and is not directly represented in the object file because it is all zero, and there is no need to put a big block of zeroes in the object file, we just need to know the starting address and length.  Because it is not directly represented in the object file, object files are smaller.  Also, it is faster to zero a block of memory at run time than to copy it from a file or rom memory.  So this makes programs smaller and faster.

It is not safe to assume that uninitialized memory will be zero, so the memory needs to be set to zero every time the program runs.

Any variable that is uinitialized or initialized to zero may be placed in bss.  Even if you do not have any uninitialized or zero-initialized variables, the start files and libraries you are linking with may have some.

There are clearly variables in bss.  Your objdump output is showing the variable names.

GCC provides the -fno-zero-initialized-in-bss option which will put zero-initialized variables into data instead of bss.  Your programs will be a little bigger, and take longer to load into memory, but you will spend less time zeroing bss at run time.  Again, you may still have zero-init variables from startfiles and libraries, so you may still have bss to clear at run-time.

nitish2112, 2019-11-07:
Thanks a lot. -fno-zero-initialized-in-bss solved the problem :-)

TommyMurphyTM1234, 2022-09-27:
Question answered.

closed 2022-09-27 by TommyMurphyTM1234