riscv-collab/riscv-gnu-toolchain #541: rv32im FPGA implementation, problems with libgcc

Ratoone-V, 2019-11-07:
Hi,

I'm quite new to RISCV ISA, but I liked the idea, that's why I decided to try to implement a basic CPU in an FPGA.

Since there, it was working pretty well, I succeeded to make a simple rv32i core, and compile C programs with standard libs using riscv-unknown-elf-gcc. I wrote a little makefile, linker script and a small piece of asm to initialize the SP.

I also implemented an UART mapped in cpu address range on the FPGA side, allowing to communicate directly with my computer. All the functions I use in my program (basic string.h functions, strtol, atoi and sprintf) do behave normally .

I was able to write simple test programs that were perfectly running.

So I decided to implement the Multiply/Divide set of instructions on the FPGA and then, things started to go wrong.

The program I try to execute is simple, it takes 4 numbers (2 int and 2 unsigned on 4 bytes) and test most of the M set instructions (_mul_, _div_, _divu_, _rem_, _remu_).

My code is working when I just compile my _"main.c"_ with option `--march=rv32im --mabi=ilp32`. I can see in the disassembly code that the _mul_, _div_, _divu_, _rem_ and _remu_ instructions are correctly generated. So i'm able to test these instructions with my PC sending data over UART.
The results are correct, even for the divide by 0 and overflow cases specified in the riscv specs.

I then tried to use the C standard libs compiled for rv32im arch. And that's when problems started to appear. Some functions like sprintf started to behave strangely. In my case, sprintf refuses to convert int or unsigned int that are >10 in strings. But if I use hexadecimal with %x, it works and displays the correct results.

The problem only appears when I use _libgcc.a_ compiled for rv32im, from _/opt/riscv/lib/gcc/riscv32-unknown-elf/9.2.0/rv32im/ilp32_. Keeping everything compiled for rv32im (my own code, libc, libg, libnosys) and just switching back to libgcc for rv32i solve the problem.

Here is my makefile:
`default: start.s main.c drivers.h drivers.c`
`	riscv32-unknown-elf-gcc -c drivers.c -o ./build/drivers.o -march=rv32im -mabi=ilp32`
`	riscv32-unknown-elf-gcc -c main.c -o ./build/main.o -march=rv32im -mabi=ilp32`
`	riscv32-unknown-elf-gcc -c start.s -o ./build/start.o -march=rv32im -mabi=ilp32`
`	riscv32-unknown-elf-ld ./build/main.o ./build/start.o ./build/drivers.o -L /opt/riscv/riscv32-unknown-elf/lib/rv32im/ilp32 -L /opt/riscv/lib/gcc/riscv32-unknown-elf/9.2.0/rv32im/ilp32 -lc -lnosys -lg -lm -lgcc -T link.ld -o ./build/main`
`	riscv32-unknown-elf-objcopy -O binary build/main test.bin`

I tried many things before this post: 
  - git pull and recompile riscv-gnu-toolchain with `--march=rv32im` and `multilib-enable`
  - use the default CPU init code by compiling all my code at once (without separate link phase) using `riscv32-unknown-elf-gcc drivers.c main.c start.s -o ./build/main -march=rv32im -mabi=ilp32 -T link.ld -mno-relax`
  - try various compiler and linker options
  - init SP at different adresses

Is it possible that there is a problem in libgcc.a ? Could it be related to the fact that my CPU starts executing code at address 0x00 ? Am I missing something ?

Thanks,

Quentin

jim-wilson, 2019-11-07:
There is no known problem with libgcc.  it sounds like there is a subtle bug in either your multiply or divide implementations.  You need to test them with more values, and try to cover all of the edge cases, such as mutltiplying two numbers that overflow, all combinations of negative and positive numbers, etc.

You could also try to debug the C library routines to see why they are failing.  That could give clues as to what is wrong with the implementations of the hardware instructions.

You could try testing multiply and divide separately.  There is a -mno-div option that you can use to disable divide while still having multiply support.  You can put this in CFLAGS_FOR_TARGET and CXXFLAGS_FOR_TARGET in the toplevel Makefile to build libraries with it, or use the new --with-target-cflags and --with-target-cxxflags configure options.  If you still see failures then multiply is broken.  If it works, then divide is broken.  It is also possible that both are broken, so check for that if you have to fix multiply.

Ratoone-V, 2019-11-07:
Thanks for this fast answer Jim, I saw a lot of your contributions and the only thing I could say is that you are truly impressive.

Yes, that's what I thought, I'm gonna investigate my implementation and try to disable division in the libs ! 

I'll let you know if I find the solution.

Ratoone-V, 2019-11-12:
Hi Jim,

I found a solution to my problem, it was indeed my division/remainder blocks that were not good.

I stopped to use Vivado's divider generator blocks and coded one myself, now my CPU is able to execute RV32IM code without any problems.

Thanks for your time and your help.

closed 2019-11-12 by Ratoone-V