riscv-collab/riscv-gnu-toolchain #542: Abouut -march=rv64IMC

SaiVK, 2019-11-08:
I have built the riscv-gnu-toolchain as part of the riscv-tools, with the options  **--with-arch=rv64imc** and **--with-abi=lp64**. On inspecting the dump of *libgcc.a*, *libm.a*, i found that it has FLD, FSD ... instructions that are part of the **F/D** extension. Can anyone explain why/how those instructions are generated even if I build the toolchain itself with those specific configuration. 

PS: The *crt.o* and cimilar object files are fine, their FLAGS indicate it is SOFT-FLOAT 
So is it that *libgcc.a* and *libm.a* some other files are not affected by these configuration options. 

Thanks in advance
-Sai

kito-cheng, 2019-11-08:
I would suggest you use `riscv64-unknown-elf-gcc -v` to check how your gcc configured.

Here is an example output, and you can found, some thing like `--with-abi=lp64d --with-arch=rv64gcv` in the output, that's the default configure for the gcc.
```
$ /home/kito/riscv-gnu-toolchain-workspace/rv64gcv/bin/riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/home/kito/riscv-gnu-toolchain-workspace/rv64gcv/bin/riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/kito/riscv-gnu-toolchain-workspace/rv64gcv/bin/../libexec/gcc/riscv64-unknown-elf/9.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/kito/riscv-gnu-toolchain-workspace/rv64gcv/build/../../riscv-gnu-toolchain-rvv/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/kito/riscv-gnu-toolchain-workspace/rv64gcv/build/.. --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/kito/riscv-gnu-toolchain-workspace/rv64gcv/build/../riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../../../riscv-gnu-toolchain-rvv/riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64gcv --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medlow' : (reconfigured) 
Thread model: single
gcc version 9.2.0 (GCC) 
```

jim-wilson, 2019-11-08:
Weird things can happen if you use a --prefix that was already used for a compiler build.  Maybe if you build a compiler for rv64gc/lp64 and then build a compiler with the same prefix for rv64imc/lp64 you can hit this problem.  I'm just guessing though, I haven't tried to reproduce this yet.  Anyways, if you aren't using a unique prefix for every compiler build, or using rm -rf to remove old compilers before building new ones, then this might be how you hit this problem.

SaiVK, 2019-11-08:
@kito-cheng  The version info specifies it as Soft-Float. As @jim-wilson specified I have used    
 **rm -rf** to remove the old hard-float toolchain and installed the soft-float in the same prefix. It is working now. 
Thank you so much for your prompt response. @kito-cheng , @jim-wilson

closed 2019-11-08 by SaiVK