riscv-collab/riscv-gnu-toolchain #543: question about adding "mod" instruction

WilliamWangPeng, 2019-11-11:
Hi dear all
    I have added the corresponding shell commands to riscv-gdb as well, and followed the four steps at
https://nitish2112.github.io/post/adding-instruction-riscv/
In the riscv-isa-sim/riscv/encoding.h add the following lines:

#define MATCH_MOD 0x200006b                                                    
#define MASK_MOD 0xfe00707f
...
DECLARE_INSN(add, MATCH_MOD, MASK_MOD)
Create a file riscv-isa-sim/riscv/insns/mod.h and add these lines:

WRITE_RD(sext_xlen(RS1 % RS2));
Add this file to riscv-isa-sim/riscv/riscv.mk.in

riscv_insn_list = \
      ...
      mod \
      ...
In riscv-isa-sim/spike_main/disasm.cc add the following lines:

DEFINE_RTYPE(mod);
and I meet with new errors like the picture one(mod.cc failed) and in riscv-isa-sim there are many .cc,.o and .d of some instructions in picture two.

if I put the shell command "spike pk mod", where mod is the elf, which will occur "an illegal instruction was executed" shown in picture two.
[image]
[image]

WilliamWangPeng, 2019-11-11:
by the way
I have already changed 
DECLARE_INSN(add, MATCH_MOD, MASK_MOD)
to 
DECLARE_INSN(mod, MATCH_MOD, MASK_MOD)

jim-wilson, 2019-11-11:
Spike isn't part of the toolchain, and the instructions you are following are out of date.  But the problem seems obvious from your screen shot.  Compare the MATCH_MOD sed command on the last line to the MATCH_AND sed command one line above.  Then figure out how to fix it.

WilliamWangPeng, 2019-11-11:
> by the way
> I have already changed
> DECLARE_INSN(add, MATCH_MOD, MASK_MOD)
> to
> DECLARE_INSN(mod, MATCH_MOD, MASK_MOD)

closed 2019-11-11 by WilliamWangPeng

WilliamWangPeng, 2019-11-11:
> Spike isn't part of the toolchain, and the instructions you are following are out of date. But the problem seems obvious from your screen shot. Compare the MATCH_MOD sed command on the last line to the MATCH_AND sed command one line above. Then figure out how to fix it.

thank you dear Jim
    I have delete "//new added" which informs me of new added instruction but made the rebuilding error, cause the compiler may misunderstand this in the rebuilding process. At the same time, I added 
#define match_mod 0x200006b                                                    
#define mask_mod 0xfe00707f
in disasm.cc. After that, there will not be errors like  
"disasm.cc ::error, "match_mod" was not declared in this scope"when  rebuilding Spike.
    finally, I put 
riscv64-unknown-elf-gcc  -o mod mod.c
spike pk mod
    Spike appears
[image]

aswaterman, 2019-11-11:
Are you sure that the spike you are executing is the same one you built?  What that dump is saying is that `02e787eb` is an illegal instruction, but in fact that pattern corresponds to the new instruction you've added.  Did you try `build/spike pk mod` from the same directory?

WilliamWangPeng, 2019-11-11:
> Are you sure that the `spike` that's in your PATH is the same one you just built? What that dump is saying is that `02e787eb` is an illegal instruction, but in fact that pattern corresponds to the new instruction you've added. So I would expect this program to execute correctly. Perhaps you forgot to `make install`, or the install path isn't in your PATH?

thank you for your advice, dear Andrew.
I use the testing program provided by Nitish
https://nitish2112.github.io/post/adding-instruction-riscv/
and now 
when I run "spike pk mod", it works.
[image]

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2