riscv-collab/riscv-gnu-toolchain #544: gcc and linux installation issue

Haonhoan, 2019-11-17:
I first installed riscv64-unknown-elf-gcc through the terminal and then tried to run this:

~$ echo -e '#include <stdio.h>\n int main(void) { int year = 2019;
> printf("Hello ECE %d!\\n", year); return 0; }' > hello.c
~$ riscv64-unknown-elf-gcc -o hello hello.c

and then I received this error:

hello.c:1:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.

Also, I tried to install RISC-V GNU Compiler Toolchain through README file and when I reached the Installation (Newlib), I get an error that says:

User:/opt/riscv/bin# ./configure --prefix=/opt/riscv
bash: ./configure: No such file or directory

I am using Ubuntu 19.10

jim-wilson, 2019-11-17:
The missing stdio.h indicates a problem with your toolchain build.  There are ways to try to debug this, using ls to look at the install tree, using gcc -v to look at the directories gcc is searching for header files, etc.  This requires learning a bit about how gcc works.

You should not be running configure in the install tree.  You run configure in the source tree.  But a single configure and make is all that is required to build the entire toolchain.  There is no extra step to build newlib.

One thing to beware of is that if you already have a riscv compiler in /opt/riscv when you try to build a toolchain, this can lead to build failures.  You should be using a unique prefix for every toolchain build, or else removing the old toolchain before trying to build a new one.

If you have a build log, there might be an error in there indicating what went wrong with the build.

If you are having trouble building a toolchain,  you could try downloading a prebuilt one.  SiFive for instance has prebuilt toolchains at
    sifive.com/boards
If you have a pre-built toolchain, you can try comparing it against the one you built to try to figure out what went wrong.

Haonhoan, 2019-11-17:
/usr/lib/gcc/riscv64-unknown-elf/8.3.0/include
 /usr/lib/gcc/riscv64-unknown-elf/8.3.0/include-fixed

I looked here and the stdio.h file is missing.

jim-wilson, 2019-11-17:
Your previous comment mentioned /opt/riscv/bin.  Now you are in /usr/lib?  Do you have more than one riscv64 compiler installed with different prefixes?

The dirs in /usr/lib/gcc are for gcc specific header files, and gcc copies of system header files, when those system header files violate ISO C rules and need to be modified, which is not common anymore.

If --prefix=/usr, then the newlib header files would be /usr/riscv64-unknown-elf/include.

I don't recommend using --prefix=/usr as that will mix the riscv compiler with system utilities.  It would be simpler to put the riscv compiler someplace separate, so that it will be easier to manage.  E.g. if you want to upgrade the compiler, and it is someplace separate, then you can just rm -rf the old one.  But you can't do that if it is installed into /usr.

Haonhoan, 2019-11-17:
I checked /opt/riscv/bin and nothing is in there. I only have it in /usr/lib. So ideally, I would install the toolchain into /opt/riscv/bin, but what do I do with the toolchain in /usr/lib? Do I just delete it?

jim-wilson, 2019-11-19:
Deleting stuff in /usr/lib is tricky.  You might accidentally delete something that the OS installed.  You probably also have riscv stuff installed into /usr/bin and /usr/libexec and /usr/share.  If you want to try deleting this stuff, I would recommend a backup first in case you make a mistake.  It is safer to just leave it there.  This stuff should be relatively harmless provided that you remember to set the path to a working riscv compiler, to avoid accidentally using a broken compiler in /usr/bin.

JunningWu, 2019-11-26:
ok，I have also got this error. 

ubuntu 16.04.2
version tag 1.0.13

# riscv64-unknown-elf-gcc  -v -o hello -c hello.c 
compiling is ok

the building log for riscv64-unknown-linux-gnu-gcc is like this:
```
/tmp/cczdhfn9.s: Assembler messages:
/tmp/cczdhfn9.s:46: Error: unrecognized opcode `ldu a5,completed.3243'
Makefile:1020: recipe for target 'crtbeginS.o' failed
make[2]: *** [crtbeginS.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/tmp/ccQy9yOd.s: Assembler messages:
/tmp/ccQy9yOd.s:5044: Error: unrecognized opcode `ldu a1,.LANCHOR0+2'
/tmp/ccQy9yOd.s:5279: Error: unrecognized opcode `ldu a5,.LANCHOR0'
/tmp/ccQy9yOd.s:5357: Error: unrecognized opcode `ldu a4,.LANCHOR0+2'
/tmp/ccQy9yOd.s:6002: Error: unrecognized opcode `ldu a2,.LANCHOR0+2'
/tmp/ccQy9yOd.s:6094: Error: unrecognized opcode `ldu a3,.LANCHOR0+2'
/home/llvm/workspace/pulp-riscv-gnu-toolchain/riscv-gcc/libgcc/static-object.mk:17: recipe for target 'unwind-dw2.o' failed
make[2]: *** [unwind-dw2.o] Error 1
make[2]: Leaving directory '/home/llvm/workspace/pulp-riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc'
Makefile:11207: recipe for target 'all-target-libgcc' failed
make[1]: *** [all-target-libgcc] Error 2
make[1]: Leaving directory '/home/llvm/workspace/pulp-riscv-gnu-toolchain/build-gcc-linux-stage1'
Makefile:150: recipe for target 'stamps/build-gcc-linux-stage1' failed
make: *** [stamps/build-gcc-linux-stage1] Error 2
```
```
llvm@llvm-vm:~/workspace/gem5/tests/test-progs/hello_haawking$ riscv64-unknown-linux-gnu-gcc  -v -o hello -c hello.c
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
Target: riscv64-unknown-linux-gnu
Configured with: /home/llvm/workspace/pulp-riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-newlib --without-headers --disable-shared --disable-threads --with-system-zlib --enable-tls --enable-languages=c --disable-libatomic --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-bootstrap --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc
Thread model: single
gcc version 7.1.1 20170509 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-o' 'hello' '-c' '-march=rv64imafdc' '-mabi=lp64d'
 /opt/riscv/libexec/gcc/riscv64-unknown-linux-gnu/7.1.1/cc1 -quiet -v hello.c -quiet -dumpbase hello.c -march=rv64imafdc -mabi=lp64d -auxbase-strip hello -version -o /tmp/ccKd2DDm.s
GNU C11 (GCC) version 7.1.1 20170509 (riscv64-unknown-linux-gnu)
	compiled by GNU C version 7.4.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring nonexistent directory "/opt/riscv/sysroot/usr/local/include"
ignoring nonexistent directory "/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/7.1.1/../../../../riscv64-unknown-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/7.1.1/include
 /opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/7.1.1/include-fixed
 /opt/riscv/sysroot/usr/include
End of search list.
GNU C11 (GCC) version 7.1.1 20170509 (riscv64-unknown-linux-gnu)
	compiled by GNU C version 7.4.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: 4919eae5429015c09459006e8f290cb7
hello.c:31:10: fatal error: stdio.h: No such file or directory
 #include <stdio.h>
          ^~~~~~~~~
compilation terminated.

```

jim-wilson, 2019-12-02:
ldu is a RV128 instruction, and there is no RV128 support in FSF GCC.  The "pulp" in your directory names is suspicious.  I don't know anything about pulp gcc versions.

I don't know what "version tag 1.0.13" means.

The stdio.h error log shows the gcc version as 7.1.1.  There are no stable FSF GCC riscv versions before 7.3.  If you built glibc, then stdio.h should be in /opt/riscv/sysroot/usr/include.  If you didn't build glibc, then you won't have stdio.h.

renjunw, 2021-04-16:
hello，i also meet this problem on ubuntu 20.
the  riscv64-unknown-elf-gcc is installed using apt-get

wangrj@ubuntu20:/opt$ riscv64-unknown-elf-gcc  -v -o hello -c hello.c
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-3seJsn/gcc-riscv64-unknown-elf-9.3.0=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
gcc version 9.3.0 ()
COLLECT_GCC_OPTIONS='-v' '-o' 'hello' '-c' '-march=rv64imafdc' '-mabi=lp64d'
 /usr/lib/gcc/riscv64-unknown-elf/9.3.0/cc1 -quiet -v -imultilib rv64imafdc/lp64d hello.c -quiet -dumpbase hello.c -march=rv64imafdc -mabi=lp64d -auxbase-strip hello -version -o /tmp/ccVgfXzK.s
GNU C17 () version 9.3.0 (riscv64-unknown-elf)
        compiled by GNU C version 9.3.0, GMP version 6.2.0, MPFR version 4.0.2, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/9.3.0/../../../../riscv64-unknown-elf/sys-include"
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/9.3.0/../../../../riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/riscv64-unknown-elf/9.3.0/include
 /usr/lib/gcc/riscv64-unknown-elf/9.3.0/include-fixed
End of search list.
GNU C17 () version 9.3.0 (riscv64-unknown-elf)
        compiled by GNU C version 9.3.0, GMP version 6.2.0, MPFR version 4.0.2, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: 4eda98260c11a5a86f1e8c1fb36a0953
hello.c:1:10: fatal error: stdio.h: No such file or directory
    1 | #include <stdio.h>
      |          ^~~~~~~~~
compilation terminated.

jim-wilson, 2021-04-16:
This is a problem with Ubuntu 20.04, not with riscv-gnu-toolchain.  Ubuntu has binutils and gcc packages for riscv64-unknown-elf, but no libraries.  You can file a bug report against Ubuntu.  Or you can use the riscv64-linux cross compiler packages instead.  Or you can download a riscv64-unknown-elf binary from somplace like the SiFive web site.  Or you can build it yourself from riscv-gnu-toolchain sources.
]

TommyMurphyTM1234, 2022-09-29:
Not an issue with the RISC-V GCC toolchain itself.

closed 2022-09-29 by TommyMurphyTM1234