riscv-collab/riscv-gnu-toolchain #547: How to resolve the problem?

SmilemanX, 2019-11-20:
/usr/bin/install: cannot overwrite directory '/home/wxuser/WX/riscv/opt/riscv/bin/openocd/openocd' with non-directory
make[3]: *** [install-binPROGRAMS] Error 1
make[2]: *** [install-am] Error 2
make[1]: *** [install-recursive] Error 1
make: *** [install] Error 2

jim-wilson, 2019-11-20:
riscv-gnu-toolchain doesn't include openocd.  I'll assume you are building riscv-tools.  But do you actually need riscv-tools?  That includes random tools that are primarily useful to hardware designers.  If you aren't trying to design your own RISC-V cpu, you probably don't need it.  riscv-tools is poorly maintained, with a difficult to use build script.  It is best to avoid it if you don't need it.

Somehow you managed to get a directory installed into the $RISCV/bin directory.  There should only be binaries in there.  I don't recall anyone reporting this error before, and I can't think of any way to do that with a normal riscv-tools build, so I can't suggest anything other than trying to debug your build.  Look for anything in the build log that is trying to install an openocd directory in $RISCV/bin.  If you have a previous failed built using the same $RISCV dir, then maybe a previous build created the bad openocd dir.  You could try deleting the bad directory in the $RISCV/bin dir and building again.  Or maybe trying a new build from scratch, either with a different RISCV value or by using rm -rf to remove the entire install tree before the build.  Though would then have to rebuild the toolchain again if you delete everything, so that would be inconvenient.

TommyMurphyTM1234, 2022-09-27:
OpenOCD question/issue. Not relevant here.

closed 2022-09-27 by TommyMurphyTM1234