riscv-collab/riscv-gnu-toolchain #550: make linux problem

emrebil, 2019-11-24:
Hello, I'm trying to compile the toolchain. After I installed those prerequiset

emrebil, 2019-11-26:
I did some more tries and now I cannot pass over the following:

[config.log](https://github.com/riscv/riscv-gnu-toolchain/files/3893979/config.log)

make -C build-gcc-linux-stage1 inhibit-libc=true all-target-libgcc
make[1]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libiberty'
make[3]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libiberty/testsuite'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libiberty/testsuite'
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libiberty'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/lto-plugin'
make  all-am
make[3]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/lto-plugin'
make[3]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/lto-plugin'
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/lto-plugin'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/intl'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/intl'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/build-x86_64-pc-linux-gnu/libiberty'
make[3]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/build-x86_64-pc-linux-gnu/libiberty'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/build-x86_64-pc-linux-gnu/fixincludes'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/build-x86_64-pc-linux-gnu/fixincludes'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/build-x86_64-pc-linux-gnu/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/build-x86_64-pc-linux-gnu/libcpp'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libbacktrace'
make  all-am
make[3]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libbacktrace'
true  DO=all multi-do # make
make[3]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libbacktrace'
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libbacktrace'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libcpp'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libdecnumber'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/libdecnumber'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/fixincludes'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/fixincludes'
make[2]: Entering directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc'
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc'
Checking multilib configuration for libgcc...
mkdir -p -- riscv32-unknown-linux-gnu/libgcc
Configuring in riscv32-unknown-linux-gnu/libgcc
configure: creating cache ./config.cache
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for --enable-version-specific-runtime-libs... no
checking for a BSD-compatible install... /usr/bin/install -c
checking for gawk... /usr/bin/gawk
checking for riscv32-unknown-linux-gnu-ar... /home/emre/opt/riscv/riscv32-unknown-linux-gnu/bin/ar
checking for riscv32-unknown-linux-gnu-lipo... riscv32-unknown-linux-gnu-lipo
checking for riscv32-unknown-linux-gnu-nm... /home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/nm
checking for riscv32-unknown-linux-gnu-ranlib... /home/emre/opt/riscv/riscv32-unknown-linux-gnu/bin/ranlib
checking for riscv32-unknown-linux-gnu-strip... /home/emre/opt/riscv/riscv32-unknown-linux-gnu/bin/strip
checking whether ln -s works... yes
checking for riscv32-unknown-linux-gnu-gcc...  /home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/home/emre/opt/riscv/riscv32-unknown-linux-gnu/bin/ -B/home/emre/opt/riscv/riscv32-unknown-linux-gnu/lib/ -isystem /home/emre/opt/riscv/riscv32-unknown-linux-gnu/include -isystem /home/emre/opt/riscv/riscv32-unknown-linux-gnu/sys-include   
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
[... cut ...]
/tmp/ccVY65Zq.s:26: Error: unrecognized opcode `sub a1,a1,a5'
/tmp/ccVY65Zq.s:27: Error: unrecognized opcode `ret'
make[2]: *** [Makefile:500: _negdi2.o] Error 1
make[2]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc'
make[1]: *** [Makefile:11267: all-target-libgcc] Error 2
make[1]: Leaving directory '/home/emre/riscv/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [Makefile:272: stamps/build-gcc-linux-stage1] Error 2

Any idea to make those opcodes recognized?

jim-wilson, 2019-12-03:
It looks like your riscv assembler is broken.  It isn't clear why.  So the very first thing you try to compile fails with an assembler error.  You can add -v to the riscv gcc command to see what assembler it is trying to use, and verify that it is actually a riscv assembler.  You can try running the riscv assembler in the install tree to see if it works.

ShraddhaDevaiya, 2019-12-16:
One more thing here, i also got similar error in linux system and i ran the build again. so after that how can we check that our build got successful or not means by which command we can ensure that riscv-gnu-toolchain is installed properly .

jim-wilson, 2019-12-16:
Try building a hello world program.

ShraddhaDevaiya, 2019-12-17:
is it anywhere available ? because i have little-bit knowledge about riscv ISA.

jim-wilson, 2019-12-17:
Just create a small C program and try to compile it.

closed 2022-09-27 by TommyMurphyTM1234