riscv-collab/riscv-gnu-toolchain #551: gmon.out generated on target board is failing due to EOF Exception while visualizing in Eclipse GPROF viewer

magicallyindia, 2019-11-29:
Hello 
We are using the riscv-gnu-toolchain and used -pg to do the profiling on the remote target board . The gmon.out is produced but its not been able to visualize in the Eclipse Linuxtools GPROF viewer plugin. 

The viewer throws an end of file exception while decoding the histogram. 

The same gmon.out was able to generate output using gprof2dot using graphviz 

Attached is the gmon.out and output from gprof2dot. 

The OS on which Eclipse is installed is Ubuntu 19.x 

Target is a RISCV board. 

[gmon.zip](https://github.com/riscv/riscv-gnu-toolchain/files/3903810/gmon.zip)

[image]

Could you please help what could be the error. 

Thanks 
Bhushan

TommyMurphyTM1234, 2022-09-27:
No update/activity in almost three years and there's nothing to suggest that this is a RISC-V GCC toolchain issue as opposed to an issue with the eclipse gprof viewer plugin which is the origin of the crash in this case.

closed 2022-09-27 by TommyMurphyTM1234

dshm, 2023-11-16:
@magicallyindia  Hello, How do you get the gmon.out file for riscv binary? As shown in the picture, the benchmark you use to do profiling is coremark. I also need to do some profiling for coremark on RISCV target. But when I compile using riscv64-unknown-elf-gcc with -pg it report errors undefined reference to _mcount' same as https://github.com/riscv-collab/riscv-gnu-toolchain/issues/771 . Have you meet same error? If so, how do you solve it?  Thank you and wait for your reply!

TommyMurphyTM1234, 2023-11-16:
> @magicallyindia Hello, How do you get the gmon.out file for riscv binary? As shown in the picture, the benchmark you use to do profiling is coremark. I also need to do some profiling for coremark on [...]

No need to spam the same question multiple times:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/695#issuecomment-1813931991
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/693#issuecomment-1813944894
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/551#issuecomment-1813957919

@jim-wilson explains the issue here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/771#issuecomment-727096961