riscv-collab/riscv-gnu-toolchain #552: Encoding of HINT instructions

abdullahyildiz, 2019-11-29:
I work on commit id 3ac0d05.

Regarding an issue at https://github.com/google/riscv-dv/issues/329#issue-529862741, I wonder that why riscv64-unknown-elf-gcc encodes HINT instructions in compressed format although an assembly program includes the `norvc` directive in it.

kito-cheng, 2019-12-02:
@Nelson1225

Nelson1225, 2019-12-02:
Dear @abdullahyildiz ,

Thanks for your report :)

User can use two methods to prevent assembler generating the RVC code.
1. Set -march=rv[32|64][extension without c-ext]
For example, -march=rv64g
2. Use the .option rvc/norvc to enable and disable RVC for the specific code region.  The priority of the .options are higher than the -march setting.

We should use .option rvc/norvc carefully, since we do meet the problem that linker may generate the RVC code in the NORVC region.  And we are working hard to resolve this problem for now.  You can see the details at the following page:
https://github.com/riscv/riscv-elf-psabi-doc/pull/116

The above problem is linker's relaxation issue, and I think your problem might not be caused by the linker.  As I know, if the ".option norvc" is set in the assembly file, then the following code will not be generated to the RVC one, and the RVC instructions are not allowed (c.xxx insn).  Except that the ".option rvc" is set again.  Beside, the .option directives are only valid for the same assembly file (.S file).   Therefore, you should set the ".option norvc" to all the files which are expected to disable the c-extension. 

So back to your question, the RVC code shouldn't be generated if we have set the ".option norvc" directive and without setting the ".option rvc" again.  This behavior is not only for the hint instructions, but also for the whole RISCV instructions I think.  Could you please provide the unexpected assembly files (.S files) to us?  If it is the assembly problem, then we can find some clues in the assembly files.  But if it is the linker problem, then only the linker relaxation patterns (lui and call relaxations) may cause this behavior, and we are trying to resolve this :)  But again, your example provided in the google/riscv-dv#329 doesn't look like a linker problem.

Best Regards and thanks
Nelson

abdullahyildiz, 2019-12-02:
@Nelson1225,

There is no `rvc` directive in the assembly file. And this is how I generate the corresponding .elf file from the assembly file:

`/opt/riscv/bin/riscv64-unknown-elf-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles out_2019-11-28/asm_tests/riscv_hint_instr_test_0.S -I/home/ayildiz/Desktop/work/riscv-dv-latest/user_extension -T/home/ayildiz/Desktop/work/riscv-dv-latest/scripts/link.ld  -o out_2019-11-28/asm_tests/riscv_hint_instr_test_0.o  -march=rv64g -mabi=lp64`

I attached the assembly file and its corresponding disassembly file that is generated by riscv64-unknown-elf-objdump.

[riscv_hint_instr_test_0.S.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3909973/riscv_hint_instr_test_0.S.txt)

[riscv_hint_instr_test_0.dump.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3909972/riscv_hint_instr_test_0.dump.txt)

kito-cheng, 2019-12-02:
Hi @abdullahyildiz 

Seems like you are using `.2byte` directive to produce hint instruction, it not compressed by compiler or assembler, it's compressed because it hard coded to compressed instruction.
```
                  lbu        a7, -139(a4)
                  sb         s2, -245(a4)
                  sra        t6, s4, s11
                  .2byte 0x704d # kHintInstr
                  lh         t0, 139(a4)
                  lb         s8, 181(a4)
                  lb         a7, -46(a4)
                  lb         a6, 88(a4) #end riscv_load_store_rand_instr_stream_0
main_15:          jal        s11, 2f
```

Nelson1225, 2019-12-02:
Dear @abdullahyildiz,

I think I find the problem.  According to the assembly file your provided, there are some instructions are written by ".2byte" directive rather than the opcode name.  The first RVC instruction I see is C.LI in the main_58_0_t, and it is written by the .2byte 0x5001 # kHintInstr.  As I know, the assembler will regards these .byte/.word/... directives as the data, rather than the instruction.  Although user can write instruction in the text sections by these data directives, assembler won't do any checking for them.  That is, if you write "li zero, -32", then assembler will generate the instruction according to the current ISA setting.

Thanks and Regards
Nelson

Nelson1225, 2019-12-02:
Yes, @kito-cheng also mentioned the same code that cause the unexpected instructions :)

abdullahyildiz, 2019-12-02:
@kito-cheng @Nelson1225,
OK. Thanks for your help. I'll report this issue.

closed 2019-12-02 by abdullahyildiz