riscv-collab/riscv-gnu-toolchain #553: Invalid memory offsets loaded with ADDI in bootcode

jeb1499, 2019-11-30:
I'm compiling a basic led blinker to verify my custom RISCV core with the default elf32lriscv.x linker.
The bootcode that is created uses negative offsets from _start to try and access higher-addressed memory locations.  I think I understand what's going wrong here.  The offset that it's trying to load with ADDI would be 12-bit, and has the last bit set, so it's being sign-extended into the register, creating a negative offset rather than a positive one.  Any idea why the compiler is choosing to use ADDI for loading a 12-bit value that should not be sign-extended?

```
000100b0 <_start>:
   100b0:       00002197                auipc   gp,0x2
   100b4:       d7018193                addi    gp,gp,-656 # 11e20 <__global_pointer$>
   100b8:       c3c18513                addi    a0,gp,-964 # 11a5c <_edata>
   100bc:       c5c18613                addi    a2,gp,-932 # 11a7c <__BSS_END__>
   100c0:       40a60633                sub     a2,a2,a0
   100c4:       00000593                li      a1,0
   100c8:       1f4000ef                jal     ra,102bc <memset>

```
p.s. Thanks for all your work on this. Seems pretty solid otherwise.

aswaterman, 2019-11-30:
The linker compensates for the addi instruction’s sign extension by
incrementing the value loaded by the lui or auipc instruction. The effect
is that the desired address is loaded into the destination register,
irrespective of the value of bit 11.

jeb1499, 2019-11-30:
Ah thanks for making me look harder at it.
I forgot that the AUIPC 0x2 is actually PC + 0x2000.
The original issue I was seeing is a result of my core's implementation of that command.
Seems like the immediate value is not propagating correctly...

closed 2019-11-30 by jeb1499