riscv-collab/riscv-gnu-toolchain #558: qemu riscv installation issue

tashfia-r, 2019-12-14:
Hi, I am trying to install qemu riscv using the following link -
https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html
At the last step in the busybear folder when I tried to run the command - make -j $(nproc)
the following error shows up -
error: riscv64-unknown-linux-gnu-gcc not found in PATH
Makefile:10: recipe for target 'busybear.bin' failed
make: *** [busybear.bin] Error 1

Please help figure out this issue.

jim-wilson, 2019-12-16:
The compiler is the first thing built in the "Building" section of that getting started guide.  You probably did something wrong there.  Check to make sure your PATH is OK and is pointing at your toolchain install tree.  Cd into your toolchain install tree and run ls in the bin dir to check is you have both the elf and linux toolchains there.  If you have your build tree, cd into that and use ls to check to see if you have both the elf and linux toolchain build dirs.  Maybe you built the elf (newlib) toolchain but forgot to build the linux toolchain?

tashfia-r, 2019-12-16:
Thanks. I resolved it.

kito-cheng, 2020-01-10:
Close issue since resolved.

closed 2020-01-10 by kito-cheng