riscv-collab/riscv-gnu-toolchain #559: Unable to compile array and C++ Code

Mohsannaeem, 2019-12-17:
Hi ! I had compiled riscv-gnu-toolchain (32I) as you described and just compiled Simple HelloWord code But when i compiled my real code full of Arrays and Vector.And alot of error arived That is Think is due to my installation orr g++ and I more thing i use c++11 in my command with Linker (.ld) and generated Map file(.map)  
[image]
Kindly Just push me to solve this Problem
Thanks In Advance ....

jim-wilson, 2019-12-17:
You used --specs=nosys.specs which disables use of libgloss, which means no system call stubs, which means you can't expect calls to read, write, etc to work unless you provide your own replacements for them.  Libgloss only has simulator support, so it is reasonable to disable it if you want code to run on hardware, but you either need to avoid making system calls like read/write/etc or else you need to provide your own versions of them.

A hello world program needs write and other system calls, so isn't expected to work on an embedded system without a BSP.

Incidentally, for SiFive boards, our BSP is found in github.com/sifive/freedom-e-sdk.  You didn't mention the board you are using, but if not a SiFive board there is probably a BSP available for it somewhere.

Mohsannaeem, 2020-01-09:
Hi once again....
I am working on my FYP project where i need to compile baremetal c++ code  on https://github.com/skordal/potato/tree/master/software  
But the problem is my code consist of dynamic allocation  and also use vector which also allocate memory using new () ..But the problem is --specs=nosys.spec  
my question is     
         Is there any way to compile  dynamic allocation..?
         Can you sugges me any material which help me compiling baremetal code and writing linker script  relevent material ?

Thanks in advance...

ilg-ul, 2020-01-09:
> Is there any way to compile dynamic allocation..?

Definitely. `new` calls `malloc()` which calls `sbrk()` which in newlib calls `_sbrk()`, that you have to implement in your application. 

Common implementations for `_sbrk()` take the entire space from the end of the bss section to the end of the RAM space, but you can implement any allocation policy.

Mohsannaeem, 2020-01-09:
Thanks 
@ilg-ul  
My linker script is 
ENTRY(_start)

MEMORY
{
	RAM (rwx)    : ORIGIN = 0x00000000, LENGTH = 0x00200000
	AEE_ROM (rx) : ORIGIN = 0xffff8000, LENGTH = 0x00400000
	AEE_RAM (rw) : ORIGIN = 0xffc00000, LENGTH = 0x00400000
}

SECTIONS
{
	.text :
	{
		*(.init)
		*(.text*)
		__text_end = .;
		*(.rodata*)
	} > AEE_ROM

	.data : AT(ADDR(.text) + SIZEOF(.text))
	{
		__data_begin = .;
		*(.data*)
		*(.eh_frame*)
		__data_end = ALIGN(4);
	} > AEE_RAM

	.bss ALIGN(4) :
	{
		__bss_begin = .;
		*(.bss*)
		*(.sbss)
		__bss_end = ALIGN(4);
	} > AEE_RAM

	/* Use the top of RAM and downwards for the stack: */
	__stack_top = 0x00000000;

	/DISCARD/ :
	{
		*(.comment)
	}
}
Can you suggest any sceme ..?
I am very newbie in linker script and don't have much knowledge yet

ilg-ul, 2020-01-09:
> I am very newbie in linker script

Then I suggest you read the documentation.

At first sight the linker script is ok, you can use `__bss_end` in `_sbrk()` as the start address in your basic allocator.

Mohsannaeem, 2020-01-09:
I read about these error and know its due to some input output functions But i am not using any cout orr cin  even i am not using #include<iostream>  
I am just using 
#include<string>
#include<vectors> 
using namespace std;

[image]
And yet I have not gotten any error due to _sbrk ().
Can you tell me does any of given include produce these errors and Why?
Thanks

ilg-ul, 2020-01-09:
Sorry, but your question does not have simple answer.

I suggest you generate C++ projects with the GNU MCU Eclipse SiFive project template and take a look at the source code there, to figure out a solution.

jim-wilson, 2020-01-12:
You mentioned C++.  You are probably using some part of libstdc++ that pulls in other code that requires system calls to be implemented.  The C++ standard library is complicated, and it is very easy to get in a situation where you need support for unexpected system calls.

You can try commenting out parts of your code to try to figure out which parts of libstdc++ are causing the trouble.

You can try adding stubs for the missing syscalls that just abort or trap or whatever, and maybe that will help you debug which parts of your code are pulling them in.  Or maybe they are never actually called, and the code will work if you stub them out.

Maybe disassemble code to see where the syscalls are coming from.  Or try generating a linux map file and look at that.  Or try adding the --verbose flag to the link to see if that gives you any useful info.

TommyMurphyTM1234, 2022-09-27:
Various questions about building C++ programs but no issues with the RISC-V GCC toolchain identified.

closed 2022-09-27 by TommyMurphyTM1234