riscv-collab/riscv-gnu-toolchain #561: Cannot open linker script file riscv.ld

Toxic-Scofield, 2019-12-27:
Hi，
    I have finished installation(Newlib). However, when I use riscv32-unknown-elf-gcc to compile an application, errors occur like follows.
[image]

  Does anybody know what the problem is here?
  Thanks!

bsousi5, 2019-12-27:
You need to specify a path to your linker file using -T option, like the following;
	riscv64-unknown-elf-gcc -o hello.elf hello.c -Tmetal.lds

Toxic-Scofield, 2019-12-27:
> You need to specify a path to your linker file using -T option, like the following;
> 	riscv64-unknown-elf-gcc -o hello.elf hello.c -Tmetal.lds
> …
> 
>  Hi，
>  I have finished installation(Newlib). However, when I use riscv32-unknown-elf-gcc to compile an application, errors occur like follows.
>   <https://user-images.githubusercontent.com/53291918/71506438-e4bfec80-28bb-11ea-9b9f-ab04239d2309.png>
>  Does anybody know what the problem is here?
>  Thanks!

email_source=notifications&email_token=AIQNVX23E4WLSMADB3YPMOLQ2WT4LA5CNFSM4J7XLTJ2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IC2DYHQ>, or unsubscribe <https://github.com/

Thanks!
Actually, I don't know which linker file I should use. When I use metal.lds, the same error occurs like this. 
[image]

jim-wilson, 2019-12-27:
I see that you are using gcc 7.1.1.  That is an old obsolete unsupported and probably broken compiler.  It isn't clear where you got this from.  Please don't use anything before gcc 7.3, which is the first version that I trust.  Current riscv-gnu-toolchain is gcc 9.x.

There is no riscv.ld in riscv-gnu-toolchain, so it isn't clear why you are looking for that file or why you can't find it.  However, this might have something to do with you using an old and obsolete compiler.  Otherwise, I'd suggest that you might have some environment variable set that is confusing the compiler to look for a file that doesn't exist.

The metal.lds that Bunnaroath suggested can be found in sifive/freedom-e-sdk which you probably aren't using, so that doesn't help you.

Oh wait, you are using pulp tools?  They are an offshoot of the berkeley tools, maintained separately.  I've never used them.  You will have to ask someone who works on the pulp tools for help.

Toxic-Scofield, 2019-12-29:
> I see that you are using gcc 7.1.1.  That is an old obsolete unsupported and probably broken compiler.  It isn't clear where you got this from.  Please don't use anything before gcc 7.3, which is [...]

Thanks！ I will check it later.

closed 2020-01-03 by Toxic-Scofield

20171166, 2020-07-14:
@Toxic-Scofield  I have a same issue. If you solve this issue, can you share your solution?

Shiva-prog123, 2020-08-06:
I too have he same issue 
[/home/sd/Desktop/RISCV/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot open linker script file metal.lds: No such file or directory
collect2: error: ld returned 1 exit status](url)
I am using RISCV pulppissimo tool chain

kito-cheng, 2020-08-06:
> I too have he same issue
> /home/sd/Desktop/RISCV/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot open linker script file  metal.lds: No such file or directory
> collect2: error: ld returned 1 exit status
> I am using RISCV pulppissimo tool chain

`metal.lds` is not provide by toolchain, it might provided by your project, maybe you could check where is `metal.lds`.

Shiva-prog123, 2020-08-06:
```
/home/sd/Desktop/RISCV/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot open linker script file riscv.ld: No such file or directory
collect2: error: ld returned 1 exit status
```

Shiva-prog123, 2020-08-06:
I think the issue is with riscv.ld linker file
Is there any way to resolve this issue

20171166, 2020-08-06:
https://stackoverflow.com/questions/62905976/linker-error-undefined-reference-to-fbss

Here is the solution that I found. Does it work for you?

Shiva-prog123, 2020-08-06:
I tried this but it's not working
same error

jim-wilson, 2020-08-06:
The riscv.ld error is not related to the metal.lds error.  I gave an answer for the riscv.ld error in #685

zhangping00, 2021-06-16:
Could you give me a hand？I can't find the riscv.ld link file when I use the RISCV toolchain in CoreMark, but it works when I use the RISCV toolchain in JTAG.  The environment variable PATH has not changed.

jim-wilson, 2021-06-16:
There is no riscv.ld file in riscv/riscv-gnu-toolchain.  But the Pulp compiler has one.  Are you maybe using the wrong compiler?  Or maybe the wrong compiler arguments?  CoreMark could be adding options that confusing the Pulp compiler.  Or maybe the "JTAG" is adding a compiler option that is missing from the CoreMark makefiles?  You can add -v to gcc to see what options are being passed to the linker.