riscv-collab/riscv-gnu-toolchain #562: Using more registers when using RV32I

lemmerelassal, 2020-01-02:
Dear RISC V team,

Thank you very much for your toolchain. I've noticed that at most 7 registers are used with ILP32: r10 to r16 (along with ra and sp). I expected it to use all possible registers, since it would help a lot when it comes to out-of-order execution.

Thank you and have a blessed 2020!

aswaterman, 2020-01-02:
The compiler will use many more registers when appropriate:

```
$ cat test.c
int foo(int* x, int* y, int n)
{
  int a = x[0];
  int b = x[1];
  int c = x[2];
  int d = x[3];
  int e = x[4];
  int f = x[5];
  int g = x[6];
  int h = x[7];
  int i = x[8];
  int j = x[9];
  int k = x[10];
  int l = x[11];
  int m = x[12];
  int o = x[13];
  int p = x[14];
  y[0] = a;
  y[1] = b;
  y[2] = c;
  y[3] = d;
  y[4] = e;
  y[5] = f;
  y[6] = g;
  y[7] = h;
  y[8] = i;
  y[9] = j;
  y[10] = k;
  y[11] = l;
  y[12] = m;
  y[13] = o;
  y[14] = p;
}
$ riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32 -O2 -S test.c -o -
	.file	"test.c"
	.option nopic
	.text
	.align	1
	.globl	foo
	.type	foo, @function
foo:
	addi	sp,sp,-16
	lw	t2,8(a0)
	lw	t0,12(a0)
	lw	t6,16(a0)
	lw	t5,20(a0)
	lw	t4,24(a0)
	lw	t3,28(a0)
	lw	t1,32(a0)
	lw	a7,36(a0)
	lw	a6,40(a0)
	lw	a2,44(a0)
	lw	a3,48(a0)
	lw	a4,52(a0)
	lw	a5,56(a0)
	sw	s0,12(sp)
	sw	s1,8(sp)
	lw	s0,4(a0)
	lw	s1,0(a0)
	sw	t2,8(a1)
	sw	s0,4(a1)
	sw	s1,0(a1)
	sw	t0,12(a1)
	sw	t6,16(a1)
	sw	t5,20(a1)
	sw	t4,24(a1)
	sw	t3,28(a1)
	sw	t1,32(a1)
	sw	a7,36(a1)
	sw	a6,40(a1)
	sw	a2,44(a1)
	sw	a3,48(a1)
	sw	a4,52(a1)
	sw	a5,56(a1)
	lw	s0,12(sp)
	lw	s1,8(sp)
	addi	sp,sp,16
	jr	ra
	.size	foo, .-foo
	.ident	"GCC: (GNU) 7.2.0"
```

TommyMurphyTM1234, 2022-09-27:
No issue here.

closed 2022-09-27 by TommyMurphyTM1234