riscv-collab/riscv-gnu-toolchain #564: Have problem with ./configure

Aquaticfuller, 2020-01-03:
When I use `./configure --prefix=/opt/riscv` in ` /opt/riscv/riscv-tools/riscv-gnu-toolchain`  with bash as the `/bin/sh`  in a Ubuntu18.04 windows subsystem, it just turns out syntax errors: 
```
./configure: line 14: $'\r': command not found
./configure: line 29: syntax error near unexpected token `newline'
'/configure: line 29: `     ;;
```
I am really confused as I have followed the README file instructions, I am really in need of help, thinks a lot !

Aquaticfuller, 2020-01-03:
I just redownloaded the file `configure` and replace the original one, and retried the `./configure --prefix=/opt/riscv`, ,it just worked. I am wondering if I just did some dumb change on that file...

TommyMurphyTM1234, 2022-09-27:
No RISC-V GCC toolchain or build process issue identified.

closed 2022-09-27 by TommyMurphyTM1234