riscv-collab/riscv-gnu-toolchain #565: GCC plugins problem

TATAyu0413, 2020-01-06:


jim-wilson, 2020-01-07:
Plugins are code that is meant to be loaded into gcc at run-time.  gcc does not run on embedded elf systems, so compiling a plugin for an embedded elf system makes no sense.  Presumably you are building a riscv cross compiler on an x86_64-linux system, which means you should be using an x86_64-linux compiler for the plugin.

The plugin support requires OS support, which linux does have, but the embedded elf/newlib compiler does not have, so what you are doing isn't expected to work.

TATAyu0413, 2020-01-08:
> Plugins are code that is meant to be loaded into gcc at run-time. gcc does not run on embedded elf systems, so compiling a plugin for an embedded elf system makes no sense. Presumably you are [...]

I want to compile my plugins that can be used by riscv64-unknown-elf-gcc and simulate using Spike.  That is to say it is not feasible.

jim-wilson, 2020-01-08:
You can build plugins for a riscv64-unknown-elf-gcc cross compiler, but this is not a riscv64 embedded elf binary, it is probably an x86_64 linux binary, hence the plugin needs to be compiled for x86_64-linux using the same compiler that was used to build the cross compiler.

TATAyu0413, 2020-01-09:
I tried to use x86_64-pc-linux-gnu-g++ that compiled riscv64-unknown-linux-gnu-g++ to compile my plugins, but when I run：
`/opt/riscv/bin/riscv64-unknown-linux-gnu-g++ -fplugin=./warn_unused.so test.cc -o test
`it ocuured errors:
```
This GCC plugin is for version 7.1
cc1plus: error: fail to initialize plugin ./warn_unused.so
```
I am grateful for your reply, it really help me a lot! But I am confused why it still ocuured errors.

jim-wilson, 2020-01-09:
This means your plugin init function failed, i.e. it returned non-zero.  You will have to debug that to see what went wrong.

TATAyu0413, 2020-03-03:
I have solved it. Thank you so much for your help!

closed 2020-03-03 by TATAyu0413

dovedevic, 2021-07-08:
@TATAyu0413 were you able to compile a file including `<sys/mman.h>` using `riscv64-unknown-elf-gcc`? If so, I am running into the same issue where my compiler is not finding this module. Any pointers?

jim-wilson, 2021-07-08:
sys/mman.h requires an OS, and embedded elf compilers assume no OS.  Use a linux compiler instead.  Or don't use mmap.