riscv-collab/riscv-gnu-toolchain #566: how can i compile C program to RISC-V in this case?

geonheec, 2020-01-07:
Hi, I want to compile my legacy C program to RISC-V binary. 
(in Ubuntu 14.04, kernel 3.13.0-24-generic, x86_64 architecture)

This C program include some headers, for example, <sys/socket.h> <netinet/in.h> <netdb.h> ... etc.

I just changed $CC in my Makefile from gcc to riscv64-unknown-elf-gcc. 
But it does not work because of the following error.
> fatal error: sys/socket.h: No such file or directory

<netinet/in.h> and <netdb.h> have same problem.

I built only "riscv-gnu-toolchain" and "riscv-tools". Should I do something more?

jim-wilson, 2020-01-07:
riscv64-unknown-elf-gcc compiles code for an embedded elf system, using newlib by default.  This is a target system without an OS, so things like sockets do not exist.

You want a riscv64-unknown-linux-gnu-gcc compiler instead to produce code for a riscv linux system.  riscv-gnu-toolchain builds an embedded elf compiler by default.  You can add --enable-linux when configuring to get a linux cross compiler build.

Note that if you have a recent enough OS, such as Ubuntu 18.04, then the distro already has support for riscv linux cross compilers, and you just need to install them.  In the ubuntu case, this would be via something like apt install gcc-riscv64-linux-gnu.

geonheec, 2020-01-08:
Thank you very much. The problem is solved.
But I'm having a new problem with using gdb, if you don't mind, please take a look at my gdb issue on riscv-isa-sim repository.

Have a nice day! Thank you again.

closed 2020-01-08 by geonheec

bhawandeepsingh, 2021-11-13:
@jim-wilson  Thanks, is it possible for you to provide a step by step sequence to generate a bin file from a c program (including steps to install any required tools) ?
I am working on a project and have a demo on Dec 1. I need to start testing asap. So, will be very thankful.
Thanks once again.

jim-wilson, 2021-11-13:
Depends on what you mean by a "bin" file.  That term is ambiguous.  If you want an ELF executable file, then you already have one.  That add_ten_nums_print file you already compiled should work in qemu user.  You can make this a little easier by using --static when compiling the program then it won't be linked against dynamic libraries, but it should still work the same.  Did you try running the program you already have?

A RISC-V toolchain works just like an x86 or arm toolchain.  If you know how to use gcc then it should work basically the same as other gcc programs that you have used.

If you ask a simple question I will answer it.  But I'm not going to write a textbook to try to help you.

bhawandeepsingh, 2021-11-24:
@jim-wilson  Thanks, I am able to generate bin file, but it seems wrong. Do I need to link any file for startup code ?

I followed the following steps - 
```
 sudo apt-get install gcc-riscv64-linux-gnu
 riscv64-linux-gnu-gcc --static add_ten_nums.c -o abcd
 riscv64-unknown-elf-elf2bin --input abcd --output abcd.bin

```

After this when I run it - Iget the error - 
```
The first instruction of core 0 has commited. Difftest enabled. 
Invalid inst 0x0000: pc = 0x0000000080000000
```
Thanks

jim-wilson, 2021-11-24:
How exactly are you running the bin file?  Note that converting a linux ELF file to bin isn't very useful, and probably won't run on any target.  A bin file like this is normally for bare metal systems, and a linux ELF assumes it is running on top of the linux kernel.  If you want a bare metal bin file, then you are better off using something like freedom-e-sdk which can produce bare metal programs.  Also, riscv-tests can produce bare metal programs.
https://github.com/riscv-software-src/riscv-tests
https://github.com/sifive/freedom-e-sdk

jim-wilson, 2021-11-24:
Another solution is to use a RTOS that comes with its own SDK, for instance Zephyr.  There is a getting started guide that talks about this.
https://risc-v-getting-started-guide.readthedocs.io/en/latest/index.html
https://github.com/riscv-admin/risc-v-getting-started-guide

jundijiujieke, 2023-06-07:
Can I compile the C code with rand function by riscv-toolchain.
But it shows can't find this function.
[image]

TommyMurphyTM1234, 2023-06-07:
> Can I compile the C code with rand function by riscv-toolchain. But it shows can't find this function. [image]

You've given no background to what you're trying to do such as:

1. What toolchain you're using, where you got it or how you configured and built it.
2. What command line and options you're using to compile your choice.
3. A reproducible test case and full instructions on how to use it.