riscv-collab/riscv-gnu-toolchain #567: Can I build gnu-toolchain in riscv linux?

geonheec, 2020-01-08:
Hi, I launched the virtual machine with qemu and riscv kernel is running on virtual machine. (I used redhat's riscv image)

However, there's no gdb in that virtual machine.
(And redhat does not have riscv gdb package....)
So I think, I need to build gdb by myself.

Is it possible to compile this gnu-toolchain in riscv kerenl-based virtual machine?
If it is not possible, is there any way to get gdb in my virtual machine?

geonheec, 2020-01-08:
+ I followed "Booting 64-bit Fedora" in https://wiki.qemu.org/Documentation/Platforms/RISCV

jim-wilson, 2020-01-08:
That is old info you are looking at.  Fedora does have gdb now, for about a year.  That rootfs you are using is about a year and a half old.  And you may also have a linux kernel version that is too old to have the gdb ptrace support.  So I would not bother continuing with what you have.

I would suggest instead:
https://fedoraproject.org/wiki/Architectures/RISC-V/Installing

geonheec, 2020-01-09:
Thank you, I tried your suggestion and there's gdb in fedora image.
But that image's ssh does not work well...
I followed the guide in https://fedoraproject.org/wiki/Architectures/RISC-V/Installing,
but the following command does not work.

> $ ssh -p 10000 -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no -o PreferredAuthentications=password -o PubkeyAuthentication=no root@localhost

I'm not sure if it's a problem with my computer or something else. 
Have you ever booted that Fedora RISC-V image and accessed it with ssh?

jim-wilson, 2020-01-09:
I do have one of these images with working ssh, but it is from about 6 months ago, so it isn't the exact same image.  If you are using the nightly images, it is possible that some of them are broken.  You could try using a different one.  They used to have a pointer to a known working image, but I don't see one in the tested images section.  If you need help from the fedora-riscv guys, they have an IRC channel on freenode where you can ask questions.  It is mentioned on the web page one level up from the installing page.

jim-wilson, 2020-01-10:
I just heard that a patch for qemu-4.2 is needed to make ssh work.
https://lists.nongnu.org/archive/html/qemu-devel/2020-01/msg01931.html
The bug report mention libquantum from SPEC, but ssh is apparently the same issue.

davidlt, 2020-01-10:
All newer images are provided via virt-builder for Fedora/RISCV. This was tested with QEMU and libvirt/QEMU at a time, but there are issues reported with QEMU 4.2 (newest release) as mentioned by @jim-wilson . I about to release new disk images (maybe this weekend after the final testing if nothing blows up).

geonheec, 2020-01-14:
Thank you very much!
It works well.

closed 2020-01-14 by geonheec