riscv-collab/riscv-gnu-toolchain #568: Unable to compile file using cross compiler because of "clang: error: unsupported option '--traditional-format' "

tymefighter, 2020-01-09:
Hi,
I have just built the repository on Mac and followed the instructions given, then went on to compile a file "hello.c" as follows: -

riscv64-unknown-elf-gcc hello.c 

But then I received an error message: -

clang: error: unsupported option '--traditional-format'

I could not find information about this "traditional format", can someone explain what the problem is and how can I solve it ?

Thanks

tymefighter, 2020-01-09:
[hello.c.zip](https://github.com/riscv/riscv-gnu-toolchain/files/4041979/hello.c.zip)

The "hello.c" file

jim-wilson, 2020-01-09:
--traditional-format is a GNU assembler option that must be used with old versions of GNU as.  Except that you aren't using an old version of GNU as.  And the error you got isn't from GNU as but rather clang.  And you aren't using clang, you are using a gcc cross compiler, so there should be no usage of any part of clang.  Something is wrong with your compiler build and it may be hard to figure out what is wrong.  I don't have a Mac, so I can't help much, except to point out that SiFive has mac binaries for a riscv cross compiler available on our web site at sifive.com/boards

If you want to try to debug your build, try adding -v to the gcc options and look at the output, which will tell you what assembler and linker are being used.  They should be the rsicv cross compiler ones.  Also, you may need to look for problems in your build tree, the config.log files may have info showing went wrong.

If you want to try alternate sources, the sifive/freedom-tools repo is what we use to build our binaries, and has support for building on a mac.  There are a number of special things that need to be done, such as using gnu versions of some tools instead of the default mac versions of them.

tymefighter, 2020-01-11:
Hi,

A professor suggested me to change the gcc compiler from the standard Mac compiler (which uses clang internally when gcc command is run) to that of homebrew's Mac compiler and then rebuild this repository.

I did so, then compiled a file using the cross-compiler, it worked.

Thanks

closed 2020-01-11 by tymefighter