riscv-collab/riscv-gnu-toolchain #570: Can official docker image be provided？

mingz2013, 2020-01-12:
Can official docker image be provided?

jim-wilson, 2020-01-12:
I don't know anything about docker.  It isn't part of the toolchain.

jim-wilson, 2020-01-12:
I should add that this project doesn't have any staff, just volunteers.  We don't have time to make binary releases for multiple targets.  We only provide sources.  I have no idea what work is necessary to make docker releases, and unlikely to ever have the time.  Maybe if someone else volunteers to do the work, but I'm also not sure if there is a way to add binary releases to this project.

mingz2013, 2020-01-19:
I have build a docker image:
https://hub.docker.com/r/mingz2013/riscv-gnu-toolchain

How to build:
https://github.com/mingz2013/riscv-gnu-toolchains-builder

It took a long time to build

then, we can use it in linux, macos, windows..etc.
- `docker pull mingz2013/riscv-gnu-toolchain`
- `docker run --rm mingz2013/riscv-gnu-toolchain ls -l /riscv/bin`
- `docker run --rm -v "$PWD"/app:/app -w /app mingz2013/riscv-gnu-toolchain /riscv/bin/riscv64-unknown-elf-gcc -o hello hello.c`
- `docker run --rm -v "$PWD"/app:/app -w /app mingz2013/riscv-gnu-toolchain /riscv/bin/riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32 -o hello hello.c`
- `docker run --rm -v  "$PWD"/app:/app -w /app mingz2013/riscv-gnu-toolchain /riscv/bin/riscv64-unknown-linux-gnu-gcc hello.c`

TommyMurphyTM1234, 2022-09-29:
No update/activity in over two years, a possible workaround/solution has been provided, and the README build instructions (https://github.com/riscv-collab/riscv-gnu-toolchain) should provide adequate information to anybody who might want to create a virtual platform (docker, VM, Vagrant file etc.) for their own purposes.

closed 2022-09-29 by TommyMurphyTM1234