riscv-collab/riscv-gnu-toolchain #571: How to compile malloc() for baremetal arcitecture?

Mohsannaeem, 2020-01-12:
Hy , I am using riscv32-unknown-elf-gcc to compile malloc() with --specs=nosys.specs flag ..I know malloc () use an OS call, but I want to add/make my system to compile even add space for heap but nothing worked for me 
Here is the output
[image]

And here is my linker script file 
ENTRY(_start)

MEMORY
{
	RAM (rwx)    : ORIGIN = 0x00000000, LENGTH = 0x00200000
	AEE_ROM (rx) : ORIGIN = 0xffff8000, LENGTH = 0x00400000
	AEE_RAM (rw) : ORIGIN = 0xffc00000, LENGTH = 0x00400000
}

SECTIONS
{
	.text :
	{
		*(.init)
		*(.text*)
		__text_end = .;
		*(.rodata*)
	} > AEE_ROM

	.data : AT(ADDR(.text) + SIZEOF(.text))
	{
		__data_begin = .;
		*(.data*)
		*(.eh_frame*)
		__data_end = ALIGN(4);
	} > AEE_RAM

	.bss ALIGN(4) :
	{
		__bss_begin = .;
		*(.bss*)
		*(.sbss)
		__bss_end = ALIGN(4);
	} > AEE_RAM

	/* Use the top of RAM and downwards for the stack: */
	__stack_top = 0x00000000;

 	_HeapStart = .;
    	__HeapStart = .;
    	. +=0x00200000 ;

	/DISCARD/ :
	{
		*(.comment)
	}
}

Can you help to compile malloc() for my bare metal application

jim-wilson, 2020-01-12:
The libnosys definition for sbrk.c expects that the linker script defines a symbol named end located at the end of the bss section, which will be used as the start address for the heap.  You can provide your own sbrk.c, or modify the default one, or add the end symbol to your linker script.

See https://github.com/riscv/riscv-newlib/blob/riscv-newlib-3.1.0/libgloss/libnosys/sbrk.c for the sbrk sources you are linking with.

You can see the default linker script with "riscv32-unknown-elf-ld --verbose" and notice how it defines the end and _end symbols.

Mohsannaeem, 2020-01-12:
Thanks Sir, Thank you very much, I just added _end in my linker script that's used default _sbrk(),An it works. Thanks once again .Can you please look this question https://github.com/riscv/riscv-gnu-toolchain/issues/559 Because I am stuck with this problem?

closed 2020-01-12 by Mohsannaeem