riscv-collab/riscv-gnu-toolchain #572: -X and --discard-locals option does not work in stripriscv

wyxin1008, 2020-01-15:
stripriscv --discard-locals our_binary_file

All temporary local symbols still exist.

wyxin1008, 2020-01-15:
when compiled our c source code file, I passed --discard-locals to ldriscv, the outout file contained the local symbols, .L0 etc.

jim-wilson, 2020-01-15:
You didn't provide a testcase to use to reproduce the problem.

Local symbols must be kept if they are used for relocations.  If you compiled with debug info, or if you compiled with shared libraries, then there may be relocations using the symbols. Otherwise, symbols like .L0 will be deleted automatically by default, so if they weren't already deleted then I would not expect that adding --discard-locals or using strip will change that.

If you are using an old binutils and/or gcc version, then this could be a bug that was fixed a while ago.  I think there was a related problem I fixed about 2 years ago.

I don't see any problem with the toolchain I am using.

wyxin1008, 2020-01-16:
thank you quick response.

I have attached a c source code example.

If compiled and linked it by ccriscv, ldriscv as following, the output binary file contained the symbols like .L0:

test.out: test.o
       ldriscv -m elf64lriscv --oformat=elf64-littleriscv -r $^ -o $@
test.o: test.c
       ccriscv -g -march=rv64imafdc -mabi=lp64d $^ -g -c -o $@

But if we used ccppc and ldppc, the output binary file did **NOT** contain the symbols like .L0
test.out: test.o
    ldppc -m elf64ppc --oformat=elf64-powerpc -r $^ -o $@
test.o: test.c
    ccppc -g -m64 -mcpu=e6500  $^ -c -o $@

I do not know why Local symbols need not be kept for PPC-based gcc, since they both are used for relocations.

wyxin1008, 2020-01-16:
example test.c file:

`void mytest()
{
    return;
}

int mystart()
{
    mytest();
    return 0;
}
`

wyxin1008, 2020-01-16:
[test.ppc.sym.txt](https://github.com/riscv/riscv-gnu-toolchain/files/4070238/test.ppc.sym.txt)
[test.riscv.sym.txt](https://github.com/riscv/riscv-gnu-toolchain/files/4070240/test.riscv.sym.txt)

jim-wilson, 2020-01-17:
The RISC-V port is using aggressive link-time relaxations to reduce code size at link time.  Because code addresses can change at link time, we must retain sufficient relocation info and local symbols to rewrite the code and the debug info as code addresses change.  The PPC port is not using link time relaxations this way, and hence does not need these extra symbols and relaxations.

The test.riscv.sym.txt file says it is for a relocatable file, so the local symbols and relocations using them must be there for linker relaxation to work.

You can get a RISC-V result similar to the PPC result if you turn off linker relaxation.  However, there is a known problem that we still emit the extra relocs for debug info even when linker relaxation is turned off.  So unfortunately, you will still see those extra symbols and relocs until this bug is fixed.
This is a bug in the binutils RISC-V port.  It is just a quality of implementation issue though, not a correctness issue, so hasn't been a high priority.

If you do turn off linker relaxation, you will find that you get larger and slower code, so in general it is not wise to do this.

TommyMurphyTM1234, 2022-09-27:
Not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234