riscv-collab/riscv-gnu-toolchain #573: Failed installation

hounaidarekik, 2020-01-15:
i run this command in /tmp: 
`curl -LO https://github.com/riscv/riscv-gnu-toolchain/archive/$RISCV_HASH.zip \
 && unzip $RISCV_HASH.zip \
 && cd riscv-gnu-toolchain-$RISCV_HASH \
 && ./configure --prefix=/opt/riscv-$RISCV_HASH \
 && make \
 && cd .. \
 && rm -rf riscv-gnu-toolchain-$RISCV_HASH $RISCV_HASH.zip`
where RISCV_HASH=9a8a0aa98571c97291702e2e283fc1056f3ce2e2

And i obtain these errors, need your help to fix all of them , Thanks
` /usr/bin/install -c -m 644 ./bfd.info '/opt/riscv-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/share/info'
/usr/bin/install: cannot remove '/opt/riscv-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/share/info/bfd.info': Permission denied
Makefile:680: recipe for target 'install-info-am' failed
make[5]: *** [install-info-am] Error 1
make[5]: Leaving directory '/tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/build-binutils-newlib/bfd/doc'
Makefile:1647: recipe for target 'install-recursive' failed
make[4]: *** [install-recursive] Error 1
make[4]: Leaving directory '/tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/build-binutils-newlib/bfd'
Makefile:1784: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/build-binutils-newlib/bfd'
Makefile:2665: recipe for target 'install-bfd' failed
make[2]: *** [install-bfd] Error 2
make[2]: Leaving directory '/tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/build-binutils-newlib'
Makefile:2154: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/build-binutils-newlib'
Makefile:179: recipe for target 'stamps/build-binutils-newlib' failed
make: *** [stamps/build-binutils-newlib] Error 2
`

jim-wilson, 2020-01-15:
This looks like more of a basic unix problem than a build problem.  Try running " ls -l /opt/riscv-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/share/info" to see why you can't remove it.  Maybe someone else owns the file and you don't have write permission?  Also check the directories in the path.  Maybe you don't have write permission to one of the directories.  This could happen if someone else already did a previous build with the same RISCV_HASH.

Taking a look at the sources you are building, I see it is a very old source tree from 2015.  I wouldn't recommend using anything before gcc-7.3.

hounaidarekik, 2020-01-17:
HI thanks for your your response,
As i am trying to install Flexpret https://github.com/pretis/flexpret, in the README is mentioned to use this version.
I fix this error but i get another one, any idea about it: 

`In file included from /tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/src/gcc/gcc/cp/except.c:1013:0:
cfns.gperf: In function ‘const char* libc_name_p(const char*, unsigned int)’:
cfns.gperf:101:1: error: ‘const char* libc_name_p(const char*, unsigned int)’ redeclared inline with ‘gnu_inline’ attribute
cfns.gperf:26:14: note: ‘const char* libc_name_p(const char*, unsigned int)’ previously declared here
cfns.gperf: At global scope:
cfns.gperf:26:14: warning: inline function ‘const char* libc_name_p(const char*, unsigned int)’ used but never defined
Makefile:1057: recipe for target 'cp/except.o' failed
make[3]: *** [cp/except.o] Error 1
make[3]: Leaving directory '/tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/build-gcc-linux-stage2/gcc'
Makefile:3971: recipe for target 'all-gcc' failed
make[2]: *** [all-gcc] Error 2
make[2]: Leaving directory '/tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/build-gcc-linux-stage2'
Makefile:858: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/build-gcc-linux-stage2'
Makefile:159: recipe for target 'stamps/build-gcc-linux-stage2' failed
make: *** [stamps/build-gcc-linux-stage2] Error 2
`

jim-wilson, 2020-01-17:
On Fri, Jan 17, 2020 at 5:37 AM hounaida <notifications@github.com> wrote:

> HI thanks for your your response,
> As i am trying to install Flexpret https://github.com/pretis/flexpret, in
> the README is mentioned to use this version.
>

OK.  But you are using really old stuff, and should expect trouble.  If
there weren't any major ISA changes since this was designed, a newer
compiler might still work, and should be easier to build.

I fix this error but i get another one, any idea about it:
> In file included from
> /tmp/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/riscv-gnu-toolchain-9a8a0aa98571c97291702e2e283fc1056f3ce2e2/src/gcc/gcc/cp/except.c:1013:0:
> cfns.gperf: In [...]

That is a bug in old gcc sources that old gcc versions would accept (maybe
with a warning?), but new gcc versions give an error.  You need to fix the
bug in the gcc sources, or use an older gcc version to compile it.  There
is a declaration and definition for the libc_name_p function, and one has
the gnu_inline attribute and one does not.  You need to add the gnu_inline
attribute to the one that is missing it.

Jim

TommyMurphyTM1234, 2022-09-29:
No update/activity in over two years and the issue seems moot anyway resulting, as it seems from @jim-wilson's comments, from the use of very old sources.

closed 2022-09-29 by TommyMurphyTM1234