riscv-collab/riscv-gnu-toolchain #574: riscv64-unknown-linux-gnu  cannot find -lstdc++and  lgcc_s

zhongpanwu, 2020-01-16:
Hi , I tried to use the Chipyard script to  install the tool-chain, but  the process is always broken by missing  lstdc++ file and lgcc_s file .

The detail is : /chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lstdc++.
/chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s

I copied these two same name from gcc and created the link file to map them by renaming it as "libstdc++.so"  (The original name is "libstdc++.so.6") and "libgcc_s.so"(The original name is "libgcc_s.so.1")

However, it generated a new error: "/riscv64-unknown-linux-gnu/bin/ld: skipping incompatible"

Does anyone can help me out about this problem? Many thanks for your any assistance .

jim-wilson, 2020-01-17:
I don't know what chipyard is.  It isn't part of the toolchain.  So I don't know why it might have failed.

But the skipping incompatible error from the linker suggests that you have a multilib enabled compiler.  This means that there isn't just one libstdc++.so but multiple copies of it, compiled for different architecture and abi variations.  You apparently copied the wrong one, or copied it into the wrong place.  You can see the list of multilibs with "riscv64-unknown-linux-gcc --print-multi-lib" which will give the directory name for it and the arch/abi options to use it.  If you want to fix your build error this way, then you need to copy all of them, and you need to copy them into the right place.

a0u, 2020-01-18:
For reference, [Chipyard](https://github.com/ucb-bar/chipyard) is an SoC design framework maintained by UC Berkeley that integrates various Chisel-based hardware blocks.  It provides a [shell script](https://github.com/ucb-bar/chipyard/blob/master/scripts/build-toolchains.sh) for building a cross-compiler toolchain and other RISC-V software infrastructure, which presumably is the subject of this issue.

@zhongpanwu: Since it appears you are building the ESP toolchain fork, please open an issue at the [esp-gnu-toolchain](https://github.com/ucb-bar/esp-gnu-toolchain) repo instead of here.  If possible, attach the `chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build*.log` files from your failed build.

I suspect that there was a problem with the build-gcc-linux-stage1 step - make sure that you start from a clean environment:
* Unset the `RISCV` environment variable (Don't source `env.sh` in chipyard)
* No other riscv64-unknown-linux-gnu tools are on your `PATH`
* Remove any previously installed versions from the destination directory (`/chipyard/esp-tools-install`)

closed 2020-02-26 by zhongpanwu