riscv-collab/riscv-gnu-toolchain #575: verilog-data-width option is missing (objcopy)

Riconec, 2020-01-17:
I tried to create hex file for readmemh in verilog for synthesys, but I found that objcopy lacks `--verilog-data-width` option and I get 16 bytes in line with spaces in between bytes, which is not correctly parsed in verilog. I decided to change it to 4 bytes in line by using `--verilog-data-width` as described here http://sourceware.org/binutils/docs-2.33.1/binutils/objcopy.html but objcopy gives errors because it is not present at all.

There is the second issue but I have to check more on that (endiannes swap when hex for verilog target is generated)

jim-wilson, 2020-01-17:
Not a RISC-V specific problem.  I would suggest reporting upstream.

SiFive has its own program for converting to verilog because of problems with the objcopy support.  You could try using that instead.
https://github.com/sifive/elf2hex

Riconec, 2020-01-17:
So lacking one of the features of objcopy (if compare with upstream) in riscv-gnu-toolchain is not a problem? It is good that there is other tools available, but why it can be done in binutils then and not here?

jim-wilson, 2020-01-17:
We do use upstream objcopy.  I don't understand your comment.  Maybe it is a versioning problem?  E.g. the feature was added upstream after we last synced with upstream?  It does look like the feature was added upstream mid last year.  Hence it might not be in your riscv-gnu-toolchain if it isn't recent enough, or if we haven't done a binutils upgrade recently enough.

Please keep in mind that riscv-gnu-toolchain is build scripts useful for newbies.  It isn't meant to be the only  way, or the best way, to build a cross compiler for RISC-V.  If you want upstream objcopy, you can build upstream objcopy.  I have limited time to maintain riscv-gnu-toolchain, so it may not always have the very latest version of everything.

Riconec, 2020-01-17:
Have you looked web page by link I added? There is option --verilog-data-width, that is not present in riscv-gnu-toolchain's objcopy

jim-wilson, 2020-01-17:
Sigh.  I can only give you the info I have already given you.  This isn't a bug, it is a versioning problem.  The feature you want only exists in binutils-2.33.  riscv-gnu-toolchain is using binutils-2.32.  If you want binutils-2.33 instead of binutils-2.32, then you are welcome to build it.  Eventually riscv-gnu-toolchain will be upgraded to binutils-2.33 or later, but we have limited time to maintain it, so it does not always have the very latest versions of everything.  If you don't like riscv-gnu-toolchain, then don't use it.  There are other alternatives like crosstool-ng, or you can build stuff by hand if you know how.

closed 2020-01-17 by jim-wilson