riscv-collab/riscv-gnu-toolchain #578: compile .asm and execute binary in shell

hicetnunc2000, 2020-01-24:
hi,

a while ago I was searching for some references so I could check out the typo of a translation from C to RISCV32 

in the other way around. I've been writting some assembly, would it be possible with unkown-elf to compile and execute an .asm binary dunno and check some outputs in the shell? just wondering about a possible workflow 

thanks

jim-wilson, 2020-01-24:
You can run programs in qemu, or in spike (riscv-isa-sim) using pk (proxy kernel), or in the gdb sim.  qemu is probably the easiest.  riscv-gnu-toolchain includes qemu for running the gcc testsuite, but it is not built by default.  You can build it with "make build-qemu".  Then you can run programs with "qemu-riscv32 a.out".

TommyMurphyTM1234, 2022-09-27:
No RISC-V GCC toolchain issue identified.