riscv-collab/riscv-gnu-toolchain #579: Linker crashes in glibc testcase ifuncdep1.o

kraj, 2020-01-27:
Attached testcase ends up crashing linker, I have just reduced it to one file for similicity so ignore the unresolved symbols since crash is same.

```
riscv64-yoe-linux/riscv64-yoe-linux-ld ifuncdep1.o

...
/mnt/b/yoe/build/tmp/work/riscv64-yoe-linux/glibc/2.30.9000-r0/recipe-sysroot-native/usr/bin/riscv64-yoe-linux/riscv64-yoe-linux-ld: BFD (GNU Binutils) 2.33.1.20191208 assertion fail ../../bfd/elfnn-riscv.c:2478
/mnt/b/yoe/build/tmp/work/riscv64-yoe-linux/glibc/2.30.9000-r0/recipe-sysroot-native/usr/bin/riscv64-yoe-linux/riscv64-yoe-linux-ld: BFD (GNU Binutils) 2.33.1.20191208 assertion fail ../../bfd/elfnn-riscv.c:2479
[1]    1768041 segmentation fault (core dumped)   ifuncdep1.o

```

jim-wilson, 2020-01-28:
It appears the problem is that allocate_dynrelocs in bfd/elfnn-riscv.c does not call _bfd_elf_allocate_ifunc_dyn_relocs to allocate space for ifunc dynamic relocs.  But only 3 ports call that function, x86, s390, and aarch64, which makes me wonder how the others work.  Maybe some of the older less actively maintained ports are doing something equivalent by hand.  Or maybe some of the other ports have the same bug.

There is a pending bug reported against FSF glibc that might be related to this linker problem.
https://sourceware.org/bugzilla/show_bug.cgi?id=24868

kraj, 2020-01-28:
@jim-wilson yes they seem to be related, I was also thinking it was due to ifuncs

jim-wilson, 2020-02-04:
This is now also
  https://sourceware.org/bugzilla/show_bug.cgi?id=25501

jim-wilson, 2020-02-04:
The RISC-V GNU toolchain has no usable ifunc support, so this is a major missing feature, not a simple bug fix.

This has now been classified as a glibc bug, since glibc has a configure check for as/ld support for ifunc before trying to use it.  This test checks to see if a simple testcase generates any relocations.  This test fails on systems that are PIE by default, because PIE requires generating a reloc for the testcase.  There is a glibc patch submitted to fix this by adding -no-pie when linking the testcase.
https://sourceware.org/ml/libc-alpha/2020-02/msg00084.html

I'm assuming that yocto/open embedded is PIE by default like Debian.

kraj, 2020-02-04:
Yes infact PIE is turned on in reference distro for yocto. I think once the glibc patch lands in master, I will backport it into yocto

jim-wilson, 2020-03-06:
A glibc patch has now been added upstream to fix the glibc configure problem.
https://sourceware.org/git/gitweb.cgi?p=glibc.git;h=87a698a21646b7ee620923ef5ffa9735471a8ddd

We still need to add the missing ifunc support to GNU binutils but that will have to wait for a volunteer to do the work.

kraj, 2020-04-07:
@jim-wilson thanks I have now added the glibc patch to Yocto

TommyMurphyTM1234, 2022-10-04:
> Attached [testcase](https://uclibc.org/~kraj/ifuncdep1.o) ends up crashing linker, I have just reduced it to one file for simplicity so ignore the unresolved symbols since the crash is the same.
> 
> ```
> riscv64-yoe-linux/riscv64-yoe-linux-ld ifuncdep1.o
> 
> ...
> /mnt/b/yoe/build/tmp/work/riscv64-yoe-linux/glibc/2.30.9000-r0/recipe-sysroot-native/usr/bin/riscv64-yoe-linux/riscv64-yoe-linux-ld: BFD (GNU Binutils) 2.33.1.20191208 assertion fail ../../bfd/elfnn-riscv.c:2478
> /mnt/b/yoe/build/tmp/work/riscv64-yoe-linux/glibc/2.30.9000-r0/recipe-sysroot-native/usr/bin/riscv64-yoe-linux/riscv64-yoe-linux-ld: BFD (GNU Binutils) 2.33.1.20191208 assertion fail ../../bfd/elfnn-riscv.c:2479
> [1]    1768041 segmentation fault (core dumped)   ifuncdep1.o
> ```

Cannot reproduce this with latest nightly build: https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.09.30

closed 2022-10-04 by TommyMurphyTM1234

kraj, 2022-10-04:
> Cannot reproduce this with latest nightly build: https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.09.30

interesting, its fixed which is good but do we know what fixed it.

TommyMurphyTM1234, 2022-10-04:
> interesting, its fixed which is good but do we know what fixed it.

Hi @kraj - unfortunately I myself don't but there seem to have been a few changes possibly relating to the issues mentioned in this thread committed upstream since the bug was reported. And probably a lot more other changes too in the intervening almost two years.

TommyMurphyTM1234, 2022-10-05:
I guess one could maybe use `git bisect` to find the change(s?) that fixed the issue?