riscv-collab/riscv-gnu-toolchain #580: No newlib for RV32 when building a multilib compiler?

nmeum, 2020-01-28:
I used this repository to build a multilib cross toolchain with newlib support as follows:

	$ git describe
	v20180629-133-g54d9e8d
	$ ./configure --prefix=/opt/riscv/ --enable-multilib --disable-linux --disable-gdb
	…
	$ make

Compiling binaries for RV64 and RV32 with -nostdlib works. Compiling
binaries, linked against newlib, for RV64 also works. What does not seem
to work is compiling RV32 binaries, linked against newlib. For example:

	$ cat > hello.c <<EOF
	#include <stdio.h>

	int main(void) {
		puts("Hello World!");
		return 0;
	}
	EOF
	$ riscv64-unknown-elf-gcc -o hello hello.c -march=rv32ima -mabi=ilp32
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/crt0.o: ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/crt0.o
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/crtbegin.o: ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/crtbegin.o
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memset.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-memset.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-puts.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-puts.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-strlen.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-strlen.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__call_atexit.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__call_atexit.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-atexit.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-atexit.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fvwrite.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fvwrite.o)
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fwalk.o): ABI is incompatible with that of the selected emulation:
	  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fwalk.o)
[... cut ...]
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__call_atexit.o): file class ELFCLASS64 incompatible with ELFCLASS32
	/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: file in wrong format
	collect2: error: ld returned 1 exit status

It seems to me that newlib et cetera was not build for RV32 or am I
missing something here? Is this intended behaviour or would you consider
this a bug?

jim-wilson, 2020-01-28:
You do have rv32 multilibs, you just don't have the exact combination you requested rv32ima/ilp32.  You can see the list of multilibs with "riscv64-unknown-elf-gcc --print-multi-lib".  There are so many possible combinations that we can't build them all by default, and we only build the "standard" subset.  You can configure a toolchain for the arch/abi combination you need with --with-arch=rv32ima --with-abi=ilp32.  Or if more ambitious, you can try adding rv32ima/ilp32 to the default list of multilibs that are built with --enable-multilib.

nmeum, 2020-01-29:
> There are so many possible combinations that we can't build them all by default, and we only build the "standard" subset. You can configure a toolchain for the arch/abi combination you need with [...]

Ah sorry, I didn't know about that. Thanks for the information. Maybe it makes sense to document this in the README? Closing this issue in the meantime.

closed 2020-01-29 by nmeum

ilg-ul, 2020-01-29:
> Maybe it makes sense to document this in the README? 

https://xpack.github.io/blog/2019/11/01/riscv-none-embed-gcc-v8-3-0-1-1-released/#supported-libraries

nmeum, 2020-01-29:
Hm? I am not using xpack.

ilg-ul, 2020-01-29:
It was an example how a more elaborate distribution may look like.