riscv-collab/riscv-gnu-toolchain #582: README improvement

ThomasHornschuh, 2020-01-30:
Hi, 
I would suggest to mention that the default newlib build instruction in the readme 
````
./configure --prefix=/opt/riscv
make
````
does not yield a multilib build. I think many people start with some RISC-V FPGA softcore, most of them are rv32i(m), and expect a toolchain which is able to target these cores. 
Multilib is mentioned on the Linux build instructions,  for novices it may not immediately clear that it also applies to newlib.

Nelson1225, 2020-01-31:
Hi @ThomasHornschuh,

It's true.  There are other configure options, like `--with-arch` and `--with-abi`, also can be set for the  newlib toolchain when you just want the specific architecture.  Also, it would be nice if we can add some descriptions for some commands of test suites (gcc, binutils, ...).

I used to read the configure file directly to find what I needed, but your are right, it would be great if we have more detailed in the README.  I can take the time to complete the README as far as I know, and then you can review the pull request.  However, It is welcome that you can modify the README through the pull request directly, if you don't mind :)

TommyMurphyTM1234, 2022-09-29:
No update/activity or PR provided in over two years.

closed 2022-09-29 by TommyMurphyTM1234