riscv-collab/riscv-gnu-toolchain #584: make linux failing because of missing file

prateek-jain93, 2020-02-05:
Hi, I am trying to install the tool on a Ubuntu machine 18.04.1. After running "make linux" command, the make exits at one point with following message : 
>>>>>
In file included from ../.././riscv-gcc/gcc/config/glibc-c.c:24:0:
../.././riscv-gcc/gcc/c-family/c-target-def.h:18:10: fatal error: ./c-family/c-target-hooks-def.h: Permission denied
 #include "c-family/c-target-hooks-def.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
../.././riscv-gcc/gcc/config/t-glibc:20: recipe for target 'glibc-c.o' failed
make[2]: *** [glibc-c.o] Error 1
>>>>>>
The file "c-target-hooks-def.h" seems to be build from ...riscv-gcc/gcc/MAKEFILE.in. Any reason why this might be failing to do so?

I am new to this, it will be appreciated if answer can be given in a simple manner.

Thanks in advance.

jim-wilson, 2020-02-05:
This does not look like a familiar problem.

Did you use ls to check to see if the file exists?  It should be in the build tree, in build-gcc-linux-stage1/gcc/c-family if the build is failing in the first gcc stage build, or -stage2 if the build is failing in the second gcc stage build.  If it does exist, check to make sure it is readable.  If the file wasn't created, then there may be an error earlier in the build log that explains why the file was not created, especially if you are using a parallel make.

prateek-jain93, 2020-02-06:
Hi, it is failing in -stage1 itself but the file is present and file permissions are rw-r--r--

[image]

The file location:
[image]

1. Any other clues to debug the failure?
2. Also, I don't have root permissions for this machine, can that somehow be the reason though path to my toolchain is local?

jim-wilson, 2020-02-08:
You don't need sudo to build gcc.

You should also check permissions for the c-family directory, and make sure you have read access to that.  You checked that the permissions for the file are OK.  But can you actually read it with more/less/vi/emacs or whatever?

You should also check that you don't have an unreadable c-target-hooks-def.h file somewhere else in your build or source trees.

You didn't show the complete compile command for the glibc-c.o file.  There should be a -I. option that should be used to find the c-family/c-target-hooks-def.h file from the build-gcc-linux-stage1 build dir.  You can add --verbose to the failing compiler command to get more info, but I'm not sure if that will tell you anything useful.

TommyMurphyTM1234, 2022-09-27:
No update/activity in over two years and no specific issues with the RISC-V GCC toolchain itself identified.

closed 2022-09-27 by TommyMurphyTM1234