riscv-collab/riscv-gnu-toolchain #588: ld: undefined reference to `__clzdi2' (stdlib disabled)

vbce, 2020-02-12:
The following code fails to link when stdlib is disabled:
```
/* Test file clz.c */

static int clz(long val)
{
	return __builtin_clzll(val);
}

int _start(void)
{
	return clz(10);
}                                               
```
The following command fails:
```
riscv64-unknown-linux-gnu-gcc -nostdlib  clz.c -o clz
riscv64-unknown-linux-gnu/bin/ld: /tmp/cc4RGuJL.o: in function `clz':
clz.c:(.text+0x14): undefined reference to `__clzdi2'
collect2: error: ld returned 1 exit status
```
This happens with the latest master branch of the repo.
Looks like a bug to me because I think no built-in function should be calling stdlib.

The RISC-V toolchain was configures as follows:
`./configure  --enable-multilib `

Similar command using x86_64 host gcc works fine:
`gcc -nostdlib  clz.c -o clz`

Thanks,
V.

vbce, 2020-02-12:
The same error happens with __builtin_ctz functions as well as __builtin_clz.

aswaterman, 2020-02-12:
It works for x86 because they have instructions for these, whereas on
RISC-V, they result in library calls.  So you have no choice but to link
against libgcc.  Try adding -lgcc to the end of the command.

aswaterman, 2020-02-12:
You're welcome!

closed 2020-02-12 by aswaterman