riscv-collab/riscv-gnu-toolchain #590: I how to install native compiler on RISC-V linux distribution which is running on RISC-V core

billasurendra, 2020-02-19:
[image]
I have developed RISC-V Linux Distribution from all opensource packages, I can able to boot Image on RISC-V core successfully. For executing some thing on Linux distribution, I have to cross compile particular program on x86 arch and I have to sent executable to board through scp command, then finally I am executing  on board. Every time I am doing this for executing some thing on booted Linux distribution. Can any one please suggest me how to install native compiler on my Linux Distribution, so that I can write a C program on my booted Linux Distribution, I can compile there itself and I can run exe file finally. I have glibc on my booted Linux Distro. Here I am attaching screenshot of my Booted Linux Distro for further clarifications. Please share if any one have idea how to install native compiler on Linux Distribution.

Thanks 
Billa

jim-wilson, 2020-02-19:
You can use the x86->riscv cross compiler to build a native riscv compiler.  There are a number of ways to do it.  This does generally require some understanding of how toolchain builds work though.  Since you already have glibc on the target, my preferred method would be to create a sysroot which contains the target /lib, /usr/lib, /usr/include and any other related directories like /lib64 and /usr/lib64 if they exist.  Copy the sysroot to the x86 machine you are using for builds.  Then use the --with-sysroot configure option with binutils and gcc.

I attached the scripts I used to build the first Fedora RISC-V Ada compiler.  You probably want to drop Ada, and use more recent binutils and gcc versions.  You build the cross binutils/gcc first and then the native (hosted) binutils/gcc.  Copy the native riscv toolchain to the target.  You might need to set some environment variables to get it to work right, like LD_LIBRARY_PATH.  Then you probably want to use the cross built native compiler to build a natively built compiler the normal way which will likely work a little better than the cross built one.

[h-binutils.txt](https://github.com/riscv/riscv-gnu-toolchain/files/4227221/h-binutils.txt)
[h-gcc.txt](https://github.com/riscv/riscv-gnu-toolchain/files/4227222/h-gcc.txt)
[x-binutils.txt](https://github.com/riscv/riscv-gnu-toolchain/files/4227223/x-binutils.txt)
[x-gcc.txt](https://github.com/riscv/riscv-gnu-toolchain/files/4227224/x-gcc.txt)

billasurendra, 2020-03-06:
Hi Jim,

 I have followed what you suggested , Now native gcc is there in my Linux
Distribution. But when I am trying to compile C program I am getting
following errors.

# gcc -I /usr/include/ -o out sample.c
/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld:
cannot find crt1.o: No such file or directory
/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld:
cannot find -lgcc_s
/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld:
cannot find -lc
/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld:
cannot find -lgcc_s
collect2: error: ld returned 1 exit status

 I did exporting LD_LIBRARY_PATH by using export
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib64/lp64d, because crt1.o is
present in this path.

Please suggest me something to resolve this issue.

Thank you
Billa

jim-wilson, 2020-03-06:
On Thu, Mar 5, 2020 at 10:48 PM Billa Surendra <notifications@github.com>
wrote:

> # gcc -I /usr/include/ -o out sample.c
>
> /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld:
> cannot find crt1.o: No such file or directory [...]

It seems that the compiler is working, you just need to configure it to
work properly on the system.  You can use "gcc -dumpspecs > specs" to
create a specs file.  Copy the specs file into the same dir that has the
cc1 binary.  Now when gcc starts it will read this specs file first.  You
can edit this specs file to add new default options.  You can add
-I/usr/include to cpp_unique_options.  You can add -L/usr/lib64/lp64d to
link_command.  And you will probably need another -L option pointing at
where the compiler files like libgcc.a have been installed.  You can use
gcc -v to see what the driver is doing, and use that info to adjust the
specs file.  Once you get a working config for this cross built compiler,
use it to build another compiler that you can install directly in /usr or
wherever you want to install it.  This final compiler won't need any hacks
to work.  Do remember to build and install binutils into the same place you
want to build and install gcc before you build and install gcc.

There might be a better way to do this, but I don't do this often enough to
know what it is.  You can try reading gcc docs for more configure options
that might give a better cross built compiler.  I know that there are other
configure options like --with-build-sysroot and
--with-native-system-header-dir but I haven't tried using them.  Or try web
searches to find FAQ that give better info on how to cross build compilers
like this.  Or just hack the compiler specs file to make it work like I
described above which is what I do.

Jim

billasurendra, 2020-03-13:
Dear Jim,

I have created specs file using *gcc -dumpspecs > specs *and copied to
appropriate place where cc1 binary is available. Following changes I did in
specs file

*cpp_*unique
*_options:*
*1. *%{!Q:-quiet} %{nostdinc*} %{C} %{CC} %{v} %{I*&F*} %{P} %*I
/usr/include*  %{MD:-MD %{!o:%b.d}%

*link_command:
2. %{!fsyntax-only:%{!c:%{!M:%{!MM:%{!E:%{!S:    %(*-L/usr/lib64/lp64d*)
%{!fno-use-linker-plugin

Changes are highlighted with bold, Please verify my specs changes because I
am getting same error after changing above changes.

error:

# gcc -o out sample.c
sample.c:1:9: fatal error: stdio.h: No such file or directory
 #include<stdio.h>
         ^~~~~~~~~
compilation terminated.

Thanks
Billa

jim-wilson, 2020-03-13:
On Thu, Mar 12, 2020 at 11:47 PM Billa Surendra <notifications@github.com>
wrote:

> # gcc -o out sample.c
> sample.c:1:9: fatal error: stdio.h: No such file or directory
> #include<stdio.h>
> ^~~~~~~~~
>

Add -v to the gcc command to see what the driver is doing.  You should see
it mentioning the specs file on the first line, and then a little later all
of the dirs it is searching for header files.  You can also use
-print-search-dirs to see the list of dirs searched for header files,
libraries, etc, but that won't show you specs file changes.  If
/usr/include is missing, then you can add it.

I see a %*I/usr/include* in your specs file.  That will pass down
-I/usr/include if the user specifies it.  If you want the compiler to
default to using -I/usr/include, then you need to include exactly that in
the cpp specs.  If you aren't sure how the specs work, you can just add
stuff and then check with gcc-v tmp.c to see what changed.

Jim

billasurendra, 2020-03-16:
Dear Jim,

I have followed what you told, still I am not getting why compilation is
not successful.  Please share if you have sample specs file.

Error:

 # gcc -v -o out sample.c
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/bin/../libexec/gcc/riscv64-unknown-linux-gnu/7.3.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: ../gcc-7.3.0/configure --host=riscv64-unknown-linux-gnu
--target=riscv64-unknown-linux-gnu
--prefix=/home/billa/MDP/DIY-6/sources/h_gcc/main
--with-sysroot=/home/billa/MDP/DIY-6/lj-os --disable-multilib
--enable-languages=c,c++ --with-arch=rv64imafd
Thread model: posix
gcc version 7.3.0 (GCC)
COLLECT_GCC_OPTIONS='-v' '-o' 'out' '-march=rv64imafd' '-mabi=lp64d'
 /usr/bin/../libexec/gcc/riscv64-unknown-linux-gnu/7.3.0/cc1 -quiet -v
-iprefix /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/ sample.c
-quiet -dumpbase sample.c -march=rv64imafd -mabi=lp64d -auxbase sample
-version -o /tmp/ccCVcrZx.s
GNU C11 (GCC) version 7.3.0 (riscv64-unknown-linux-gnu)
compiled by GNU C version 7.3.0, GMP version 6.1.2, MPFR version 4.0.1, MPC
version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=41 --param ggc-min-heapsize=22468
ignoring nonexistent directory
"/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/include"
ignoring duplicate directory
"/usr/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/include"
ignoring nonexistent directory
"/home/billa/MDP/DIY-6/lj-os/usr/local/include"
ignoring duplicate directory
"/usr/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/include-fixed"
ignoring nonexistent directory
"/usr/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/include"
ignoring nonexistent directory "/home/billa/MDP/DIY-6/lj-os/usr/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/include
 /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/include-fixed
End of search list.
GNU C11 (GCC) version 7.3.0 (riscv64-unknown-linux-gnu)
compiled by GNU C version 7.3.0, GMP version 6.1.2, MPFR version 4.0.1, MPC
version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=41 --param ggc-min-heapsize=22468
Compiler executable checksum: 731aa66652b218307d1174cf58b8af6b
sample.c:1:9: fatal error: stdio.h: No such file or directory
 #include<stdio.h>
         ^~~~~~~~~
compilation terminated.

Thanks

Billa Surendra

jim-wilson, 2020-03-16:
On Mon, Mar 16, 2020 at 3:18 AM Billa Surendra <notifications@github.com>
wrote:

> I have followed what you told, still I am not getting why compilation is
> not successful. Please share if you have sample specs file.j

I don't have a sample specs file.  I only did this canadian cross process
for RISC-V once 2 years ago, and now build everthing natively.

> ignoring nonexistent directory
>
> "/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/include"
> ignoring duplicate directory [...]

I would think that adding -I/usr/include to the specs file would work.  But
looking at the list of dirs searched, I see that it is looking for
/home/billa/MDP/DIY-6/lj-os/usr/include, so another possible solution is to
create the missing directories up through usr, and then make include a
symlink pointing at /usr/include.  This is probably the dir name used on
the x86-linux machine you used to build the canadian cross compiler.  This
should make the compiler happy, and then once this compiler is happy you
can use it to build a proper native compiler that looks in the right dirs.
Another possible solution is to add --sysroot=/ to tell the compiler to use
stuff in / instead of the host x86-linux sysroot.  If that works, you
should be able to put --sysroot=/ in the specs file so that you don't have
to manually specify it.

There are comments in the gcc/gcc.c source file that explain how specs
files work.

There is a -print-sysroot option to gcc that will tell you the built-in
sysroot.  Normally this will be empty, but for your canadian cross built
compiler it may be pointing at the sysroot dir on the x86-linux host
machine.

Jim

billasurendra, 2020-03-17:
> On Mon, Mar 16, 2020 at 3:18 AM Billa Surendra ***@***.***> wrote: I have followed what you told, still I am not getting why compilation is not successful. Please share if you have sample specs file.j
> I don't have a sample specs file. I only did this canadian cross process for RISC-V once 2 years ago, and now build everthing natively.
> ignoring nonexistent directory "/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/include" ignoring duplicate directory "/usr/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/include" ignoring nonexistent directory "/home/billa/MDP/DIY-6/lj-os/usr/local/include" ignoring duplicate directory "/usr/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/include-fixed" ignoring nonexistent directory "/usr/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/../../../../riscv64-unknown-linux-gnu/include" ignoring nonexistent directory "/home/billa/MDP/DIY-6/lj-os/usr/include" #include "..." search starts here: #include <...> search starts here: /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/include /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.0/include-fixed End of search list.
> I would think that adding -I/usr/include to the specs file would work. But looking at the list of dirs searched, I see that it is looking for /home/billa/MDP/DIY-6/lj-os/usr/include, so another possible solution is to create the missing directories up through usr, and then make include a symlink pointing at /usr/include. This is probably the dir name used on the x86-linux machine you used to build the canadian cross compiler. This should make the compiler happy, and then once this compiler is happy you can use it to build a proper native compiler that looks in the right dirs. Another possible solution is to add --sysroot=/ to tell the compiler to use stuff in / instead of the host x86-linux sysroot. 

your correct jim when I am typing **gcc -print-sysroot** command in Linux distribution it is displaying result is x86(host pc) path. your suggesting me to add --sysroot=/ , I am not getting what exactly I have to do can you please explain me in details about adding --sysroo=/ option to compiler.

If that works, you should be able to put --sysroot=/ in the specs file so that you don't have to manually specify it. There are comments in the gcc/gcc.c source file that explain how specs files work. There is a -print-sysroot option to gcc that will tell you the built-in sysroot. Normally this will be empty, but for your canadian cross built compiler it may be pointing at the sysroot dir on the x86-linux host machine. Jim

jim-wilson, 2020-03-17:
On Tue, Mar 17, 2020 at 2:25 AM Billa Surendra <notifications@github.com>
wrote:

> your correct jim when I am typing *gcc -print-sysroot* command in Linux
> distribution it is displaying result is x86(host pc) path. your suggesting
> me to add --sysroot=/ , I am not getting what exactly I have to do can you
> please explain me in details about adding --sysroo=/ option to compiler.
>

First check to see if  manually specifying --sysroot=/ works.  Try a hello
world program for example.

if that works, then one option is to create a specs file and add the
--sysroot=/ option into it.  You can create a specs file with -dumpspecs,
and then copy it into the dir where the cc1 binary is located.  You can
verify the specs file is being used with -v, it will print a line saying it
is using specs from your specs file.  then just add the string --sysroot=/
to an appropriate place in the specs file, like cpp_options or
cpp_unique_options, and probably also link_options.

To learn how specs files work, just start making random changes to the
specs file, and run the compiler to see what it did.  It shouldn't be hard
to figure out.  If you need more info, see the gcc.c file.

If you run into problems doing that, then another way to solve this problem
is to just create a symbolic link at the dir -print-sysroot is already
using that points back to /.

Jim

billasurendra, 2021-12-02:
Hi Jim,

I am trying to install gettext-0.21 on RISC-V (RV64IMAFD) OS image which has been developed by me. I have a native compiler on it. Here I am facing a problem while installing gettext on top of it. Other than gettext package, I can able to install other packages. Can you please help me out?

Configure command I have used is
**./configure --prefix=/usr --build=riscv64-unknown-linux-gnu**

But when I am running **make -j8** , I am getting the following error.

Error:
libtool: link: rm -fr  .libs/libtextstyle.so.0.1.1 .libs/libtextstyle.ver
libtool: link: echo "{ global:" > .libs/libtextstyle.ver
libtool: link:  cat libtextstyle.sym | sed -e "s/\(.*\)/\1;/" >> .libs/libtextstyle.ver
libtool: link:  echo "local: *; };" >> .libs/libtextstyle.ver
libtool: link:  gcc -shared  -fPIC -DPIC  .libs/gl_array_list.o .libs/basename-lgpl.o .libs/binary-io.o .libs/c-ctype.o .libs/c-strcasecmp.o .libs/c-strncasecmp.o .libs/cloexec.o .libs/concat-filename.o .libs/exitfail.o .libs/fatal-signal.o .libs/fd-hook.o .libs/fd-ostream.o .libs/file-ostream.o .libs/full-write.o .libs/get_ppid_of.o .libs/get_progname_of.o .libs/getprogname.o .libs/html-ostream.o .libs/html-styled-ostream.o .libs/iconv-ostream.o .libs/gl_list.o glthread/.libs/lock.o .libs/malloca.o .libs/math.o .libs/mem-hash-map.o .libs/memory-ostream.o .libs/noop-styled-ostream.o .libs/ostream.o .libs/printf-frexp.o .libs/printf-frexpl.o .libs/safe-read.o .libs/safe-write.o .libs/sig-handler.o .libs/sockets.o .libs/stat-time.o .libs/styled-ostream.o .libs/sys_socket.o .libs/term-ostream.o .libs/term-style-control.o .libs/term-styled-ostream.o glthread/.libs/threadlib.o .libs/unistd.o unistr/.libs/u8-mbtouc.o unistr/.libs/u8-mbtouc-aux.o .libs/xmalloc.o .libs/xstrdup.o .libs/xconcat-filename.o .libs/xgethostname.o .libs/gl_xlist.o .libs/xsize.o .libs/xvasprintf.o .libs/xasprintf.o .libs/color.o .libs/misc.o .libs/version.o .libs/asnprintf.o .libs/fcntl.o .libs/obstack.o .libs/printf-args.o .libs/printf-parse.o .libs/vasnprintf.o  -Wl,--whole-archive ./.libs/libcroco_rpl.a ./.libs/libglib_rpl.a ./.libs/libxml_rpl.a -Wl,--no-whole-archive  -lm -L/usr/lib -lncurses  -g -O2   -Wl,-soname -Wl,libtextstyle.so.0 -Wl,-version-script -Wl,.libs/libtextstyle.ver -o .libs/libtextstyle.so.0.1.1
ld[20067]: unhandled signal 11 code 0x1 at 0x0000000000000088 in ld[10000+14b000]
CPU: 1 PID: 20067 Comm: ld Not tainted 5.4.1 #1
sepc: 000000000005333c ra : 0000000000054290 sp : 0000003fffa86540
 gp : 000000000015e4d0 tp : 0000003fe083c7b0 t0 : 0000000000084cac
 t1 : 00000000000003fc t2 : 0000000000000000 s0 : 0000000000162de0
 s1 : 0000000000000000 a0 : 0000000000182b80 a1 : 0000000000000000
 a2 : 0000000000d8f0d0 a3 : 000000000000cbf4 a4 : 0000000000000003
 a5 : 0000000000149b18 a6 : 00011f0000000000 a7 : 0000000000000003
 s2 : 0000000000d8dcd0 s3 : 0000000000d86760 s4 : 00000000013a5370
 s5 : 0000000000184f20 s6 : 0000003fffa865f8 s7 : 0000000000000000
 s8 : 0000000000d980a0 s9 : 0000000000d8f0d0 s10: 0000000000d914a7
 s11: 0000000000000000 t3 : 00000000000000ff t4 : 0000000000000000
 t5 : 0000000000000001 t6 : 000000000006e1b0
sstatus: 8000000000006020 sbadaddr: 0000000000000088 scause: 000000000000000d
collect2: fatal error: ld terminated with signal 11 [Segmentation fault]
compilation terminated.
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_setup.o): relocation R_RISCV_HI20 against `TABSIZE' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_ti.o): relocation R_RISCV_HI20 against `cur_term' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_tparm.o): relocation R_RISCV_HI20 against `_nc_prescreen' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_tputs.o): relocation R_RISCV_HI20 against `stdout@@GLIBC_2.27' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_ttyflags.o): relocation R_RISCV_HI20 against `_nc_prescreen' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(name_match.o): relocation R_RISCV_HI20 against `_nc_globals' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(read_entry.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(alloc_ttype.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(comp_error.o): relocation R_RISCV_HI20 against `_nc_globals' can not be used when making a shared object; recompile with -fPIC
[... cut ...]
make[3]: *** [Makefile:1786: all-recursive] Error 1
make[3]: Leaving directory '/home/gettext-0.21/libtextstyle'
make[2]: *** [Makefile:1714: all] Error 2
make[2]: Leaving directory '/home/gettext-0.21/libtextstyle'
make[1]: *** [Makefile:403: all-recursive] Error 1
make[1]: Leaving directory '/home/gettext-0.21'
make: *** [Makefile:359: all] Error 2
/home/gettext-0.21 # ./configure --prefix=/usr --disable-static --docdir=/usr/sh
are/doc/gettext-0.21--build=riscv64-unknown-linux-gnu && make -j8 && make install

Thanks
Billa Surendra

cmuellner, 2021-12-02:
Your problem is, that you are linking your PIC code against a static library (`libncurses.a`), that has been built without `-fPIC`.
So you are trying to add non-position-independent code to build a shared library, which needs to be position-independent.

Depending on your use case I see several solutions:
* configure gettext with `--disable-curses`
* build libncurses.a on your own (ensure it is compiled with `-fPIC`) and link against this libncurses.a when building gettextpo.so
* build libgettext only as static library

Regarding the `SIGSEGV`: hard to say what's the reason for that.

Slightly off-topic: GCC 7.3 has been released in Jan 2018. Glibc 2.27 was released at the same time. That's almost four years. You might consider using more recent components.

jim-wilson, 2021-12-02:
Or build a libncurses.so.

You have the ld binary, and the kernel message gives you a register dump, so you can try loading it into a debugger to see where and why it failed.  I can't do much otherwise unless I can reproduce it, and the problem might be specific to your environment which I don't have.

billasurendra, 2021-12-03:
Thanks Jim and cmuellner, gettext installed sucessufully with --disable-curses

billasurendra, 2021-12-23:
Hi All,
I am getting the same error when  I am trying to compile libedit-20210216-3.1 on RISC-V Linux. But this  --disable-curses not working. Please suggest other alternatives.

Error:
make[3]: Entering directory '/home/package_building/libedit-20210216-3.1/src'
  CCLD     libedit.la
ld[25923]: unhandled signal 11 code 0x1 at 0x0000000000000088 in ld[10000+14b000]
CPU: 2 PID: 25923 Comm: ld Not tainted 5.4.1 #1
sepc: 000000000005333c ra : 0000000000054290 sp : 0000003ffff659f0
 gp : 000000000015e4d0 tp : 0000003fc81a67b0 t0 : 000000000003910c
 t1 : 00000000000003fc t2 : 0000000000000000 s0 : 0000000000162de0
 s1 : 0000000000000000 a0 : 000000000017a680 a1 : 0000000000000000
 a2 : 00000000004bc3b0 a3 : 000000000000828c a4 : 0000000000000003
 a5 : 0000000000149b18 a6 : 00011f0000000000 a7 : 0000000000000003
 s2 : 00000000004bafb0 s3 : 00000000004b43f0 s4 : 0000000000a41160
 s5 : 000000000017ca10 s6 : 0000003ffff65aa8 s7 : 0000000000000000
 s8 : 00000000004c4390 s9 : 00000000004bc3b0 s10: 00000000004be787
 s11: 0000000000000000 t3 : 00000000000000ff t4 : 0000000000000000
 t5 : 0000000000000001 t6 : 00000000000265a8
sstatus: 8000000000006020 sbadaddr: 0000000000000088 scause: 000000000000000d
collect2: fatal error: ld terminated with signal 11 [Segmentation fault]
compilation terminated.
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_termcap.o): relocation R_RISCV_HI20 against `_nc_globals' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_tparm.o): relocation R_RISCV_HI20 against `_nc_prescreen' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_tputs.o): relocation R_RISCV_HI20 against `stdout@@GLIBC_2.27' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(comp_error.o): relocation R_RISCV_HI20 against `_nc_globals' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(comp_userdefs.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_baudrate.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_cur_term.o): relocation R_RISCV_HI20 against `ospeed' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_napms.o): relocation R_RISCV_HI20 against `SP' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_setup.o): relocation R_RISCV_HI20 against `TABSIZE' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_ti.o): relocation R_RISCV_HI20 against `cur_term' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_ttyflags.o): relocation R_RISCV_HI20 against `_nc_prescreen' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(name_match.o): relocation R_RISCV_HI20 against `_nc_globals' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(read_entry.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(alloc_ttype.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(comp_captab.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(db_iterator.o): relocation R_RISCV_HI20 against `_nc_globals' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(entries.o): relocation R_RISCV_HI20 against `_nc_head' can not be used when making a shared object; recompile with -fPIC
[... cut ...]
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: /usr/lib/libncurses.a(lib_tparm.o)(.text+0x2440): unresolvable R_RISCV_CALL relocation against symbol `sprintf@@GLIBC_2.27'
make[3]: *** [Makefile:454: libedit.la] Error 1
make[3]: Leaving directory '/home/package_building/libedit-20210216-3.1/src'
make[2]: *** [Makefile:384: all] Error 2
make[2]: Leaving directory '/home/package_building/libedit-20210216-3.1/src'
make[1]: *** [Makefile:456: all-recursive] Error 1
make[1]: Leaving directory '/home/package_building/libedit-20210216-3.1'
make: *** [Makefile:365: all] Error 2

Thanks 
Billa

jim-wilson, 2021-12-23:
Looks like libedit is a package for editing command lines.  I seriously doubt it can be built without a curses library.  Which means that you either need to build libncurses as a shared library, or build libedit without shared library support.  The former is probably the better solution, as you are going to keep having this same problem over and over again if you continue to build more packages that need a curses library.

billasurendra, 2022-01-05:
> Looks like libedit is a package for editing command lines. I seriously doubt it can be built without a curses library. Which means that you either need to build libncurses as a shared library, or [...]

Thanks it has worked after installing ncurses package.

billasurendra, 2022-01-06:
Hi Everyone

I am trying to rebuild dnf SRPM package on my RISC-V Linux distro, with the help of rpmbuild but it is giving an error. not only dnf SRPM for every SRPM when I am stated rebuilding its giving error like this. Can anyone please help me regarding this?
Error:

/home/SRPM_Rebuilding/dnf-4.6.1-1.fc34.src # ls
dnf-4.6.1.tar.gz  dnf.spec
/home/SRPM_Rebuilding/dnf-4.6.1-1.fc34.src # **rpmbuild -ba dnf.spec** 
error: /home/SRPM_Rebuilding/dnf-4.6.1-1.fc34.src/dnf.spec: line 188: <lua> scriptlet support not built in

In line 188 below content is there
**%autosetup -p1**  

Thanks
Billa

jim-wilson, 2022-01-06:
Looks like you need to rebuild rpmbuild with scriptlet support.  There might have been a dependency missing when you built it the first time that prevented this feature from being enabled.  Check the configure output, find the missing dependency, build it, and then rebuild rpmbuild.  Or maybe there is a special configure option you have to supply to enable scriptlet support, and it isn't turned on by default.  Check the configure script or configure script docs.

billasurendra, 2022-01-12:
> Looks like you need to rebuild rpmbuild with scriptlet support. There might have been a dependency missing when you built it the first time that prevented this feature from being enabled. Check the [...]

Thanks after re-installing rpmbuild with LUA package support it is working fine

billasurendra, 2022-01-12:
Hi Jim,
I am trying to install curl-7.65.3 package from source code on my Linux distro image, but it's giving an error. openssl installed on os and also I did export. Can you please help me out?

**Error:**

make[3]: Leaving directory '/home/package_building/curl-7.65.3/docs/cmdline-opts'
make[2]: Leaving directory '/home/package_building/curl-7.65.3/docs'
make[2]: Entering directory '/home/package_building/curl-7.65.3/src'
  CCLD     curl
/usr/lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/ld: warning: libdl.so.2, needed by /usr/local/lib/libssl.so, not found (try using -rpath or -rpath-link)
/usr/local/lib/libcrypto.so: undefined reference to `dlclose@GLIBC_2.27'
/usr/local/lib/libcrypto.so: undefined reference to `dlerror@GLIBC_2.27'
/usr/local/lib/libcrypto.so: undefined reference to `dladdr@GLIBC_2.27'
/usr/local/lib/libcrypto.so: undefined reference to `dlopen@GLIBC_2.27'
/usr/local/lib/libcrypto.so: undefined reference to `dlsym@GLIBC_2.27'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:922: curl] Error 1
make[2]: Leaving directory '/home/package_building/curl-7.65.3/src'
make[1]: *** [Makefile:1966: all-recursive] Error 1
make[1]: Leaving directory '/home/package_building/curl-7.65.3/src'
make: *** [Makefile:922: all-recursive] Error 1
/home/package_building/curl-7.65.3 # cd /usr/local/lib
**/usr/local/lib # ls**
engines-1.1        liblua.a           libncurses.so      libssl.a
libcrypto.a        libmenu.a          libncurses.so.6    libssl.so
libcrypto.so       libmenu.so         libncurses.so.6.1  libssl.so.1.1
libcrypto.so.1.1   libmenu.so.6       libncurses_g.a     lua
libform.a          libmenu.so.6.1     libpanel.a         pkgconfig
libform.so         libmenu_g.a        libpanel.so        terminfo
libform.so.6       libncurses++.a     libpanel.so.6
libform.so.6.1     libncurses++_g.a   libpanel.so.6.1
libform_g.a        libncurses.a       libpanel_g.a
/usr/local/lib # ln -s libssl.so.1.1 libssl.so
ln: libssl.so: File exists
/usr/local/lib # cd
/ # cd home/package_building/curl-7.65.3/
/home/package_building/curl-7.65.3 # ./configure --prefix=/usr --with-openssl=/u
sr/local/ssl^C
/home/package_building/curl-7.65.3 # echo $PATH
/usr/local/ssl/bin:/bin:/sbin:/usr/bin:/usr/sbin:/usr/lib:/usr/local/bin

Thanks
Billa

jim-wilson, 2022-01-13:
You are missing libdl.  That should come from glibc.  Check system dirs to see if you have libdl or not.  Maybe you have it, but it is an older version?  Sometimes the ABI changes and the library version number gets bumped.  Or maybe you didn't build glibc with shared library support?

billasurendra, 2022-01-13:
Can I install latest version glibc now

billasurendra, 2022-01-18:
Hi Jim,
I have tried installing glibc-2.33 version but it's not compiling. Can you please suggest to me what to do now? actually, I am stuck here.

jim-wilson, 2022-01-18:
I'm not sure if installing a new glibc solves the problem.  You have executables looking for the libdl from an older glibc version.  There is still the question of what happened to libdl from the older glibc build.  Rebuilding the old glibc version, with libdl, is probably a simpler solution.

You didn't mention why the new glibc doesn't compile.  But I think building a new glibc will require rebuilding everything else to link with it, so you probably don't want to do this unless you have to.

billasurendra, 2022-01-20:
The command which I have used for building glibc in my Linux distribution.
Command:
BUILD_CC="gcc" CC="${LJOS_TARGET}-gcc" \
AR="${LJOS_TARGET}-ar" \
RANLIB="${LJOS_TARGET}-ranlib" CFLAGS="-O2" \
../glibc-2.27/configure --prefix=/usr \
--host=${LJOS_TARGET} --build=${LJOS_HOST} \
--disable-profile --enable-add-ons --with-tls \
--enable-kernel=2.6.32 --with-__thread \
--with-binutils=${LJOS}/cross-tools/bin \
--with-headers=${LJOS}/usr/include \
--cache-file=config.cache

Thanks 
Billa

billasurendra, 2022-02-03:
After creating a libdl.so.2 file in /usr/lib directory problem has solved. Thanks for all your support.

billasurendra, 2022-02-21:
Hi Jim,

I am trying to install cmake package on RISC-V Linux but it is not compiling and also I am unable to find out the reason. Can you please help to identify the mistake?. Here I am attaching the screenshot of the 
[image]
error message.
Thanks
Billa

jim-wilson, 2022-03-07:
Do you have more than one glibc version on this machine?  The libpthread_nonshared.a file is present in older glibc versions, but not current glibc versions.  Looks like libpthread_nonshared.a was dropped March 2018.  So maybe the problem is that you had an old glibc version originally which contained the libpthread_nonshared.a file.  Then you upgraded to a newer glibc version that does not have the libpthread_nonshared.a file but did not remove the libpthread_nonshared.a file.  Maybe the configure script is getting confused, seeing the libpthread_nonshared.a and thinking it needs to link with it, and failing.  In this case, the solution would be to remove the libpthread_nonshared.a file and reconfigure.

I'm just guessing based on inomplete info though.  There could be something else obscure going on here.

closed 2022-09-29 by TommyMurphyTM1234