riscv-collab/riscv-gnu-toolchain #594: Is it possible to get gcc/10.0.0 for RISC-V? If so, how?

dareyouspam, 2020-02-25:
I'm trying to replicate the environment, which apparently includes GCC 10.0.0, that was used by the EmBench folks to report results for their reference platform.
"_It uses a GCC tool chain built from top of tree, version 10.0.0 20190527, commit 4f90058758f._"
from https://github.com/embench/embench-iot/tree/master/doc#reference-platform

jim-wilson, 2020-02-25:
There are multiple ways to build a toolchain.  You can build direct from FSF sources if you know how.  Just build binutils first, and then build gcc+newlib.  This is what Embecosm probably did.  You can use crosstool-ng to build a toolchain, though I don't know offhand if it can build from top of tree.

You can also do this with riscv-gnu-toolchain with a little git magic.  You can use git remote to add the FSF git trees, and then check them out inside riscv-gnu-toolchain to use the FSF sources with the riscv-gnu-toolchain Makefile.  The process would be something like
  ... clone riscv-gnu-toolchain ...
  cd riscv-binutils
  git remote add upstream git://sourceware.org/git/binutils-gdb.git
  git fetch upstream
  git checkout upstream/master
  cd ../riscv-gcc
  git remote add upstream git://gcc.gnu.org/git/gcc.git
  git fetch upstream
  git checkout upstream/master
  cd ..
  ... build riscv-gnu-toolchain ...
just remember not to update any of the submodules after doing this, as that will revert them back to the original submodules, though you can use the git checkout commands again to get back to the FSF trees.

If you want the same compiler as Embecosm used, then use the git commit id you have above instead of upstream/master in the checkout command.  Generally, if you upgrade gcc you should also upgrade binutils, but if they didn't mention the binutils commit then you may not be able to exactly reproduce their environment.  Maybe just choose a binutils commit id from the same date as the gcc commit id and that should get you close enough.  Note that this still may not give the exact same compiler if they configured their compiler differently, or used a different newlib version, or used different support library versions (gmp/mpc/mpfr), etc.

dareyouspam, 2020-02-25:
That worked, thank you Jim!

closed 2020-02-25 by dareyouspam