riscv-collab/riscv-gnu-toolchain #595: please implement -msoft-float

dsw, 2020-02-26:
When I try to build using -msoft-float, I get this:

`riscv64-unknown-elf-gcc: error: unrecognized command line option '-msoft-float'`

I need to be able to build tests to generate both kinds of executables: those that use the hardware floating point ops and those that do not.  While I could reconfigure gcc to omit the RISCV extensions  F, D, and Q and then re-build it, I then have to have two copies of the compiler lying around and every time someone re-installs our system, they have to build the compiler twice.  So it would help if you could implement this flag, please.

aswaterman, 2020-02-26:
The `-m` options are target-specific, and different targets tackle this problem in different ways.  For RISC-V, we chose to orthogonalize the choice of ISA from the choice of calling convention as much as possible, and so you need to pass two arguments to pull that off, e.g., `-march=rv64imac -mabi=lp64` for soft-float or `-march=rv64gc -mabi=lp64d` for hard-float.

The multilib system understands some of these options, so I believe if you build the toolchain with `--enable-multilib`, you'll get a single toolchain that can handle either set of options.

closed 2020-03-17 by aswaterman

dsw, 2020-09-10:
Trying your suggestion, and also linking
against dewdrop-berkeley-softfloat-3, I get this message when I try to link:
"can't link hard-float modules with soft-float modules".

Do I need to rebuild the compiler to never emit hard floating point?