riscv-collab/riscv-gnu-toolchain #596: objdump does not always prefix hex operands with 0x thus emitting illegal instruction source

johnwinans, 2020-02-29:
given:

    winans@ux410ua:~/riscv-gnu-toolchain$ git describe --long --dirty
    v20180629-147-g6d2706f

Configured like this:

    ./configure --prefix=/home/winans/projects/riscv/install/rv32i --with-arch=rv32i --with-abi=ilp32

Then compile a test program like this:

    riscv32-unknown-elf-gcc -ffreestanding -fno-pic -march=rv32i -mabi=ilp32  -ffreestanding -fno-pic -march=rv32i -mabi=ilp32 -nostdlib -Wl,-Ttext=0x0 -Wl,--no-relax  badhex.S -o badhex

Where badhex.S starts with this:

            .text
            .align  2
            .globl  _start
    _start:
            jal             x1,1f 
    1:
            jal             x1,1f
            ebreak
    1:
            jal             x1,tjal1
            jal             x1,tjal1
            jal             x0,1f
    tjal1:
            jalr    x0,0(x1)                # return to caller
    1:
            ...

...and then run:

    riscv32-unknown-elf-objdump -Mnumeric,no-aliases -dr badhex

I get this:

Disassembly of section .text:

    00000000 <_start>:
       0:   004000ef                jal     x1,4 <_start+0x4>
       4:   008000ef                jal     x1,c <_start+0xc>
       8:   00100073                ebreak
       c:   00c000ef                jal     x1,18 <tjal1>
      10:   008000ef                jal     x1,18 <tjal1>
      14:   0080006f                jal     x0,1c <tjal1+0x4>

    00000018 <tjal1>:
      18:   00008067                jalr    x0,0(x1)
      ...

We can see that objdump is generating hex literals without the leading 0x.  Poking around a bit more this seems to appear on jal as well as all the branch instructions. I would expect that object would render legal source. 

I would be willing to poke around to see if I can fix it.  This would be my first reach into any of the gnu toolchain source.  Anyone care to aim me in the right direction?

Or, maybe it is a trivial fix for someone else?

Thanks,
-John

jim-wilson, 2020-02-29:
We can't accept patches for GNU projects without an FSF copyright assignment, unless they are trivial and hence not copyrightable.  If you are serious about contributing, you should try to get the paperwork in place if you don't already have one.  You can get the process started by sending email to assign@gnu.org and ask about a copyright assignment.  Generally you need an assignment from your employer, or alternatively a disclaimer from your employer and a personal assignment.  If you are a college student, you probably need a disclaimer from your college.

Objdump sources are in binutils/objdump.c, and the disassembler support in in opcodes/riscv-dis.c and disassembler.c.  See for instance riscv_disassemble_insn in the riscv-dis.c file.

johnwinans, 2020-03-02:
I am an instructor at Northern Illinois University.  I will pursue the assignment.  
Fortunately, this appears to be a trivial fix. I'd like to send a pull request for a one line addition.
I am a bit disoriented, however, by the how the riscv-binutils-gdb subproject is linked.
I am looking at this:
    winans@ux410ua:~/projects/riscv/riscv-gnu-toolchain/riscv-binutils$ git diff
    diff --git a/binutils/objdump.c b/binutils/objdump.c
    index 60a3967129..4dd95b854c 100644
    --- a/binutils/objdump.c
    +++ b/binutils/objdump.c
    @@ -1394,6 +1394,7 @@ objdump_print_addr (bfd_vma vma,
       if (!skip_find)
         sym = find_symbol_for_address (vma, inf, NULL);
     
    +  (*inf->fprintf_func) (inf->stream, "0x");
       objdump_print_addr_with_sym (aux->abfd, inf->section, sym, vma, inf,
                                   skip_zeroes);
     }

When I did the clone on riscv-gnu-toolchain I got the riscv-binutils-2.34 branch.  Should I send a against that branch or should I do it against the head of the riscv-binutils-gdb?  ...or should I be looking elsewhere upstream?
Thanks,
-John

jim-wilson, 2020-03-02:
Patches must be sent upstream.  We do all development work in the FSF source tree, and then pull stuff down into the github.com/riscv trees.  Generally, patches will only be accepted for the top of tree sources, unless it is fixing a regressions from a previous release, or a show stopper bug in a currently supported release.

See
    sourceware.org/binutils
and binutils@sourceware.org is the primary mailing list for patches and dscussions.

TommyMurphyTM1234, 2022-09-29:
No update/activity in over two years and, as far as I can tell, this is no longer a problem:

* https://sourceware.org/git/?p=binutils-gdb.git;a=blob;f=binutils/objdump.c;h=6610906f83eb330d6f8f6ce5610119c07dabca39;hb=HEAD#l1655

closed 2022-09-29 by TommyMurphyTM1234