riscv-collab/riscv-gnu-toolchain #599: Error assertion failed, when using Linux-elf/glibc toolchain

timfaner, 2020-03-08:
I build Linux-elf/glibc toolchain on Ubuntu 18.04.2,  (GNU/Linux 4.15.0-45-generic x86_64). then I use `riscv64-unknown-linux-gnu-gcc` compiler a blank c code like
```c
int main(){ return 0;}
```
Then I run elf file using `riscv64-unknown-linux-gnu-run` and I get an error:
> sim-core.c:424: assertion failed - (addr & (nr_bytes - 1)) == 0

I have no idea why this happened.

jim-wilson, 2020-03-08:
run is the gdb simuilator, which is primarily intended for embedded elf programs.  To use it for linux programs, you probably have to statically link, and maybe do other simplifying stuff.

You are better off using qemu instead.  qemu is faster and better supported and fully supports linux programs.  riscv-gnu-toolchain doesn't build qemu by default.  You can build it with "make build-qemu".  Then use qemu-riscv64 to run.  We only have qemu for running the gcc testsuite.  We don't update it often.  If you need a better qemu, grab a release from the upstream sources at qemu.org.

Long term, I would suggest getting both gcc and qemu from the OS.  Ubuntu 18.04 does have a gcc-8 riscv64-linux cross compiler package, but it doesn't have a qemu riscv64 package.  Ubuntu 20.04 hopefully will have both.

timfaner, 2020-03-09:
Thank you for your reply~

closed 2020-03-09 by timfaner