riscv-collab/riscv-gnu-toolchain #601: riscv64-unknown-elf-gcc not compille cpp file then #include <iostream>

mpak2, 2020-03-14:
1 #include <cstdio>
  2 #include <functional>
  3 
  4 #include <iostream> // TODO
  5 int main(int argc, char *argv[]){
  6    int a = 0;
  7    std::function<void()> f_display_42 = []() { printf("42"); };
  8    do{
  9       printf("Проверка %d\n", a);
 10    }while(1000 > a++);
 11    return 0;
 12 }

riscv64-unknown-elf-gcc -lstdc++ -std=c++14 -lm main.cpp -v -o main
can mistake

Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/9.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /usr/src/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv/toolchain --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/toolchain/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
gcc version 9.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-std=c++14' '-v' '-o' 'main' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d'
 /opt/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/9.2.0/cc1plus -quiet -v main.cpp -quiet -dumpbase main.cpp -mtune=rocket -march=rv64imafdc -mabi=lp64d -auxbase main -std=c++14 -version -o /tmp/ccYuX1jq.s
GNU C++14 (GCC) version 9.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 9.2.1 20191008, GMP version 6.1.2, MPFR version 4.0.2, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/opt/riscv/toolchain/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/opt/riscv/toolchain/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/include/c++/9.2.0
 /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/include/c++/9.2.0/riscv64-unknown-elf
 /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/include/c++/9.2.0/backward
 /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/include
 /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/include-fixed
 /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C++14 (GCC) version 9.2.0 (riscv64-unknown-elf)
	compiled by GNU C version 9.2.1 20191008, GMP version 6.1.2, MPFR version 4.0.2, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 2b0d21df5fa862dcbcb919a6c0379f32
COLLECT_GCC_OPTIONS='-std=c++14' '-v' '-o' 'main' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d'
 /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/as -v --traditional-format -march=rv64imafdc -mabi=lp64d -o /tmp/ccmtvXjr.o /tmp/ccYuX1jq.s
GNU ассемблер, версия 2.34 (riscv64-unknown-elf); используется BFD версии (GNU Binutils) 2.34
COMPILER_PATH=/opt/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/9.2.0/:/opt/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/9.2.0/:/opt/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/:/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/:/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/:/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/:/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/:/opt/riscv/toolchain/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-std=c++14' '-v' '-o' 'main' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d'
 /opt/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/9.2.0/collect2 -plugin /opt/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/9.2.0/liblto_plugin.so -plugin-opt=/opt/riscv/toolchain/libexec/gcc/riscv64-unknown-elf/9.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccOg81To.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgloss -plugin-opt=-pass-through=-lgcc --sysroot=/opt/riscv/toolchain/riscv64-unknown-elf -melf64lriscv -o main /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/crt0.o /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/crtbegin.o -L/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0 -L/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib -L/opt/riscv/toolchain/riscv64-unknown-elf/lib -lstdc++ -lm /tmp/ccmtvXjr.o -lgcc --start-group -lc -lgloss --end-group -lgcc /opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/crtend.o
/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccmtvXjr.o: в функции «.L0 »:
main.cpp:(.text+0x3ce): неопределённая ссылка на «std::ios_base::Init::Init()»
/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: main.cpp:(.text+0x3de): неопределённая ссылка на «std::ios_base::Init::~Init()»
/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: main.cpp:(.text+0x3e2): неопределённая ссылка на «std::ios_base::Init::~Init()»
/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccmtvXjr.o: в функции «.LC1»:
main.cpp:(.rodata+0x30): неопределённая ссылка на «vtable for __cxxabiv1::__class_type_info»
/opt/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccmtvXjr.o:(.sdata.DW.ref.__gxx_personality_v0[DW.ref.__gxx_personality_v0]+0x0): неопределённая ссылка на «__gxx_personality_v0»
collect2: error: ld returned 1 exit status

jim-wilson, 2020-03-14:
You must put the -l options at the end.  A static library is only used if needed, and it won't be needed if it is the first thing passed to the linker.  Or alternatively use g++ instead of gcc, which will handle the stdc++ and math libraries automatically.

So use this:
riscv64-unknown-elf-g++ -std=c++14 main.cpp -o main
or this
riscv64-unknown-elf-gcc -std=c++14 main.cpp -o main -lstdc++ -lm

mpak2, 2020-03-14:
Thanks a lot. I appreciate your help.

closed 2020-03-14 by jim-wilson