riscv-collab/riscv-gnu-toolchain #602: libgcc not generating soft float

sathya281, 2020-03-17:
Hi,

I have generated a toolchain with march=rv64imafdc mabi=lp64d

Now I compile programs with march=rv64imac and mabi=lp64. I get softfloat hardfloat mismatch errors. Looks like soft float instructions are not generated for libgcc. Is there anyway, i can generated soft float for libgcc ?

aswaterman, 2020-03-17:
It sounds like you generated a toolchain without multilib support.  You can either configure the toolchain with `--enable-multilib` or build a second toolchain with the alternate -march/-mabi settings.

sathya281, 2020-03-17:
Thanks Andrew

sathya281, 2020-03-17:
I generated a multilib and then triedt
 rv64imafdc libraries are getting linked ?
Is there any command I need to give explicitly ?

/home/rise/tools/riscv64/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/rise/tools/riscv64/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/rv64imafdc/lp64d/libm.a(lib_a-w_pow.o): can't link double-float modules with soft-float modules

jim-wilson, 2020-03-17:
riscv64-unknown-elf-gcc --print-multi-lib
will show you the multilib combinations you have available.  It isn't practical to build every possible combination by default, but rv64imac/lp64 is one of the default multilibs and should be there.

Just showing us one line of the output doesn't give enough info for us to help.  We need to have enough info to reproduce the problem in order to see what is wrong.

You can get more info from the compiler by adding -v if you want to try to better understand what the compiler is doing.

kito-cheng, 2020-10-11:
Muilt

closed 2020-10-11 by kito-cheng