riscv-collab/riscv-gnu-toolchain #604: make linux error: The riscv64 is not supported.

ALLap97, 2020-03-19:
I build the Newlib cross-compiler successfully。
**but i build the Linux cross-compiler with some errors.**
error seems to occur when building **stamps/build-glibc-linux-headers**.

config.log in build-glibc-linux-headers is here.

It was created by GNU C Library configure (see version.h), which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ /home/peng/Documents/riscv-gnu-toolchain/riscv-glibc/configure --host=riscv64-unknown-linux-gnu --prefix=/media/sf_both/riscv/sysroot/usr --enable-shared --with-headers=/home/peng/Documents/riscv-gnu-toolchain/linux-headers/include --disable-multilib --enable-kernel=3.0.0

## --------- ##
## Platform. ##
## --------- ##

hostname = peng-VirtualBox
uname -m = x86_64
uname -r = 4.15.0-45-generic
uname -s = Linux
uname -v = #48~16.04.1-Ubuntu SMP Tue Jan 29 18:03:48 UTC 2019

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /media/sf_both/riscv/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games
PATH: /usr/local/games
PATH: /snap/bin
PATH: /media/sf_both/riscv/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2206: checking build system type
configure:2220: result: x86_64-pc-linux-gnu
configure:2240: checking host system type
configure:2253: result: riscv64-unknown-linux-gnu
configure:2282: checking for riscv64-unknown-linux-gnu-gcc
configure:2309: result: riscv64-unknown-linux-gnu-gcc
configure:2578: checking for C compiler version
configure:2587: riscv64-unknown-linux-gnu-gcc --version >&5
riscv64-unknown-linux-gnu-gcc (GCC) 10.0.1 20200311 (experimental)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2598: $? = 0
configure:2587: riscv64-unknown-linux-gnu-gcc -v >&5
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/media/sf_both/riscv/libexec/gcc/riscv64-unknown-linux-gnu/10.0.1/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/peng/Documents/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/media/sf_both/riscv --with-sysroot=/media/sf_both/riscv/sysroot --with-newlib --without-headers --disable-shared --disable-threads --with-system-zlib --enable-tls --enable-languages=c --disable-libatomic --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-bootstrap --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
[... 113 lines cut ...]
configure:3894: riscv64-unknown-linux-gnu-gcc -g -O2  -Werror -fstack-protector-all -xc /dev/null -S -o /dev/null
configure:3897: $? = 0
configure:3905: result: yes
configure:3948: checking for assembler and linker STT_GNU_IFUNC support
riscv64-unknown-linux-gnu-gcc: fatal error: '-fuse-linker-plugin', but liblto_plugin.so not found
compilation terminated.
configure:3979: result: no
configure:3983: checking for gcc attribute ifunc support
conftest.c:10:24: error: 'ifunc' is not supported on this target
   10 | extern __typeof (func) func __attribute__ ((ifunc ("resolver")));
      |                        ^~~~
configure:4009: result: no
configure:4013: checking if compiler warns about alias for function with incompatible types
conftest.c:9:34: error: 'foo_alias' alias between functions of incompatible types 'int(const void *, int)' and 'int (*(void))(const void *, int)' [-Werror=attribute-alias=]
    9 | extern __typeof (__redirect_foo) foo_alias __attribute__ ((alias ("foo")));
      |                                  ^~~~~~~~~
conftest.c:4:28: note: aliased declaration here
    4 | __typeof (__redirect_foo) *foo_impl (void)
      |                            ^~~~~~~~
cc1: all warnings being treated as errors
configure:4035: result: yes
configure:4050: WARNING: gcc emits a warning for alias between functions of incompatible types
configure:4055: WARNING: Multi-arch is disabled.
configure:4073: checking sysdep dirs
configure:4181: error: The riscv64 is not supported.

ALLap97, 2020-03-19:
when i  use 
`./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d`
`
make linux
`
to build the 32-bit RV32GC toolchain。
the error becomes **The riscv32 is not supported.**

jim-wilson, 2020-03-25:
I haven't seen this error before.  Looking at the configure script, it looks like the loop that tries to find the correct sysdep dirs in the glibc source tree.  If you add --enable-debug-configure then you will see debugging output on stdout (not in config.log), that will look like this
DEBUG: try sysdeps/riscv/rv64/rvd/fpu/unix/sysv/linux-gnu/riscv/rv64/rvd/fpu
it goes through a list of hundreds of possible names.  If it finds at least one dir that matches, then it should succeed.  if it finds no matches, then you get the error saying that riscv64 is not supported.  So something is wrong in the list of names being generated when you run configure.

I would guess that there might be a versioning problem.  E.g. maybe you have an old OS install that has old versions of sed or whatever that don't work with the glibc configure script.  Or maybe you have a bleeding edge OS install that has new versions of sed or whatever that don't work with the glibc configure script.

Another issue is that you should see something like
hecking for sysdeps preconfigure fragments... aarch64 alpha arm csky hppa i386\
 m68k microblaze mips nios2 powerpc riscv s390 sh sparc x86_64
and it is the sysdeps/riscv/preconfigure file that sets base_machine to riscv and machine to riscv/rv$xlen/$float_machine.  $machine is printed in the error message you see, and it should be riscv/rv64/fpu instead of riscv64.  This suggests that sysdeps/riscv/preconfigure is not being found and run during configure, or else is being run but failing to work.  Maybe put some debug echo commands in sysdeps/riscv/preconfigure to see if it is being run and working correctly.

jim-wilson, 2020-04-13:
Assuming that this is the same issue as 617, then the problem is that you checked out the wrong version of riscv-glibc.  The master branch of riscv-glibc, which is the default branch, is not usable.  If you check out the master branch of riscv-gnu-toolchain, and then update submodules, you will get the correct version of riscv-glibc.

ALLap97, 2020-04-15:
you are right!my problem is the same as issue 617.
I clone the master branch of riscv-glibc before, and it's the wrong version.
then i  update submodule riscv-gilbc in riscv-gnu-toolchain, and now i can make linux.
thank you jim!

closed 2020-05-04 by kito-cheng