riscv-collab/riscv-gnu-toolchain #606: building binutils for riscv32 failed

gyh1997127, 2020-03-26:
Hi all, I am building a riscv32 cross compiler from scratch, and got the error message 

> ../../../bfd/doc/bfd.texi: Value too large for defined data type

when running  `make all -j8` for binutils. So I traced back and found 

> checking for riscv32-unknown-elf-cc... no
> checking for riscv32-unknown-elf-gcc... no
> checking for riscv32-unknown-elf-c++... no
> checking for riscv32-unknown-elf-g++... no
> checking for riscv32-unknown-elf-cxx... no
> checking for riscv32-unknown-elf-gxx... no
> checking for riscv32-unknown-elf-gcc... no
> checking for riscv32-unknown-elf-gfortran... no
> checking for riscv32-unknown-elf-gccgo... no
> checking for riscv32-unknown-elf-ar... no
> checking for riscv32-unknown-elf-as... no
> checking for riscv32-unknown-elf-dlltool... no
> checking for riscv32-unknown-elf-ld... no
> checking for riscv32-unknown-elf-lipo... no
> checking for riscv32-unknown-elf-nm... no
> checking for riscv32-unknown-elf-objcopy... no
> checking for riscv32-unknown-elf-objdump... no
> checking for riscv32-unknown-elf-ranlib... no
> checking for riscv32-unknown-elf-readelf... no
> checking for riscv32-unknown-elf-strip... no
> checking for riscv32-unknown-elf-windres... no
> checking for riscv32-unknown-elf-windmc... no

when I ran  `../configure`; Is this a problem with my current gcc? This worked before on my home ubuntu 18.04, did not work on redhat el7. Would appreciate some feedback! Thanks!

jim-wilson, 2020-03-27:
I'm not familiar with that error message, but RHEL7 has a lot of old tools.  It was originally released in 2014, which is a very long time ago in the linux world.  This might be a versioning problem with texinfo.  You need a newer OS, or a newer version of texinfo on your current OS.  "makeinfo --version" will give you the texinfo version number.

Red Hat does provide some packages, like the Red Hat Developer Toolset
https://developers.redhat.com/products/developertoolset/overview
and the Red Hat Software Collections
https://developers.redhat.com/products/softwarecollections/overview
which provide a way to get optional newer versions of tools.  Using this stuff might solve your problem if it contains a newer version of texinfo.

The missing programs is not a problem because you haven't built them yet.

gyh1997127, 2020-03-27:
Not sure why but when I ran the commands in bash instead of csh, everything worked again.

closed 2020-03-27 by gyh1997127