riscv-collab/riscv-gnu-toolchain #609: What is the compile time identifier of the gcc and g++ compilers?

Ravenwater, 2020-03-27:
We are porting a tensor algebra library that supports long doubles. This environment specifies the long double as 16bytes, different from ARM, MSVC, Intel, IBM, PCG, x86 gcc and clang. What is the compile time identifier that source code can use to specialize for this tool chain?

jim-wilson, 2020-03-27:
On RISC-V, long double is always 128-bits, so we don't need or have any predefined macro for that.  You could just check for __riscv.  In the future we may change the size of long double for a new embedded ABI, in which case we may need a predefined macro at that time.  Probably something to document the ABI, and then you can infer long double size from the ABI.

Meanwhile, you can always use sizeof (long double).  Or you can include float.h and check one of the macros defined by that file like LDBL_MANT_DIG.  Or if this is gcc, you can use one of the predefined macros used by float.h, like __LDBL_MANT_DIG which works without including float.h but is non-standard.

With gcc, you can see all of the predefined macros with
  touch tmp.c
  gcc -dM -E tmp.c

Ravenwater, 2020-03-27:
@jim-wilson you are awesome. Love that gcc -dM -E trick.

we use the compiler identifier and sizeof(long double) to specialize our code between the different implementations of long double. 

I needed that __riscv identifier so that we can specialize for this tool chain.

TommyMurphyTM1234, 2022-09-27:
Question answered.

closed 2022-09-27 by TommyMurphyTM1234