riscv-collab/riscv-gnu-toolchain #610: embedding .string into a .S file is not handled properly

MarcKarasek, 2020-03-28:
I am writing some assembly and need to embed a string for uart output.  
The code is:
	jal     print
	nop		  
	j	1f	  
	nop		  
	.string	"\n\rPort to RISC-V Hondo: Marc Karasek"
1:	nop

What is output from gcc (this is a .S file) is:
       c:	1af000ef          	jal	ra,9ba <.L2^B2+0x27>
			c: R_RISCV_JAL	print
      10:	00000013          	nop
      14:	02c0006f          	j	40 <_start+0x40>
			14: R_RISCV_JAL	.L1^B1
      18:	00000013          	nop
      1c:	0d0a                	slli	s10,s10,0x2
      1e:	6f50                	ld	a2,152(a4)
      20:	7472                	ld	s0,312(sp)
      22:	7420                	ld	s0,104(s0)
      24:	4952206f          	j	22cb8 <print_done+0x21343>
      28:	562d4353          	0x562d4353
      2c:	4820                	lw	s0,80(s0)
      2e:	6f646e6f          	jal	t3,46724 <print_done+0x44daf>
      32:	203a                	fld	ft0,392(sp)
      34:	614d                	addi	sp,sp,176
      36:	6372                	ld	t1,280(sp)
      38:	4b20                	lw	s0,80(a4)
      3a:	7261                	lui	tp,0xffff8
      3c:	7361                	lui	t1,0xffff8
      3e:	6b65                	lui	s6,0x19
	...
0000000000000041 <.L1^B1>:
      41:	00000013          	nop

The string is being appended with a 0x00 (NULL) char by the compiler but the  j 1f in hte code is not being set properly.  So it wants to jump to 0x40 offset and the location is 0x41

This generates an error in compile when it goes to link

umon.o: in function `_start':
(.text.init+0x14): relocation truncated to fit: R_RISCV_JAL against `.L1^B1'
collect2: error: ld returned 1 exit status

The disassembly is from the .o file.  So the .S -> .o compile completes it fails when it goes to link.

MarcKarasek, 2020-03-30:
I have put in a work around.   
Adding the strings at the end of the code with there own tags 
stringfoo:
.string "foo"

And then doing a la <register> to access the string seems to work.

Nelson1225, 2020-03-31:
Hi,

This is caused when you put the data into the text section.  For the current toolchain, it is better not to put the data and text together.  I remember Kuan-Lin Chen had sent a patch last year which can resolve the similar issue for .p2align/.align, but it still be discussing until now.  Your case is what I have mentioned before, we should deal with not only the .p2align/.align but also the case that put data into text.  I can only find the one of mails since there are too many...

https://sourceware.org/legacy-ml/binutils/2019-12/msg00024.html

Thanks
Nelson

MarcKarasek, 2020-03-31:
The data is still in the .text section.  It is just not embedded in the code.

The case that fails is like this
#define prstr(str)  \
	jal     print;\
	nop		  ;\
	j	1f	  ;\
	nop		  ;\
	.string	str 	  ;\
1:	nop
Here the string being used in the macro is inline with the code.  What I saw was that the j 1f along with the jal print was off by 1 byte.  The only think I could see was that the strings added NULL was not being taken into account when compiling (this is a .S file).  

If i take the strings and create a tag for them at the end of the code:
stringfoo:
.string "foo"

and then access the strings : 

	la t2, stringfoo
1:	lb	a0,0(t2)
	beq	a0,zero,2f
	jal umon_conout
	addi t2, t2,1
	j	1b
2:
 
This at least compiles and looks ok in objdump.  I have not had a chance to test the code yet but will in a few days. 

Marc

TommyMurphyTM1234, 2022-09-29:
No update/activity in over two years and insufficient info provided to easily identify a reproducible test case and a specific issue with the RISC-V GCC toolchain.

closed 2022-09-29 by TommyMurphyTM1234