riscv-collab/riscv-gnu-toolchain #611: How to recompile GCC without fully rebuilding it?

save-buffer, 2020-03-29:
Hi, 
I am currently modifying riscv-gcc's optimizer to know about a new architecture. So far I've been building riscv-gcc using this toolchain, and then deleting `stamps/build-gcc-newlib-stage2` and then rerunning `make` to recompile gcc. 
Unfortunately this removes all of the existing build artifacts and does a clean rebuild. Is there a way to reduce compile times by avoiding a rebuild from scratch?

Thanks!

jim-wilson, 2020-03-29:
cd build-gcc-newlib-stage2
make 
make install

That is effectively what the makefile does when you rm stamps/build-gcc-newlib-stage2, except without the rm -rf and configure.

save-buffer, 2020-03-29:
Thanks for your reply! This seems to work - I'd tried it before and it was telling me it didn't know the option `-no-pie`. I updated my version of g++ to a newer version.

closed 2020-03-29 by save-buffer