riscv-collab/riscv-gnu-toolchain #612: If --program-prefix argument to configure isn't supported, what's its alternative?

musamaanjum, 2020-03-30:
Currently, the built toolchain has the prefix "riscv64-unknown-elf-". I want to rename the prefix to "riscv64-uet-". But I don't see --program-prefix="uet" is supported by the configure tool even though it is fork of the upstream gnu toolchain. If this option isn't supported, please suggest how can I build the toolchain with my prefix?

More details about --program-prefix can be obtained from https://gcc.gnu.org/install/configure.html

jim-wilson, 2020-03-30:
Every part of the toolchain supports --program-prefix.  it is just the toplevel makefile which is riscv-gnu-toolchain specific that doesn't support it as no one ever added it.  You can modify the makefile to pass --program-prefix down to the submakes.  You could also try changing the make_tuple makefile variable to what you want.

musamaanjum, 2020-03-30:
I've achieved my intended purpose by editing make_tuple in Makefile.in. Thanks

closed 2020-03-30 by musamaanjum