riscv-collab/riscv-gnu-toolchain #613: compiling .S program results in weird errors: undefined reference to `ra'

MarcKarasek, 2020-03-30:
Code is 👍 
mv 	a0,s1		/* echo the range */
jal printreg

It does not happen for all jal but a small subset.. And seems to travel if you comment out the jal that is causing the error:

The error is always the same:
(.text.init+0x1514): undefined reference to `ra'

This seems to indicate it could not resolve the register ra used to store the return address from the jal??

jim-wilson, 2020-03-30:
It appears that you have a reference to a symbol called ra instead of a reference to the register ra.  Try running nm on your object file.  Does it report a symbol called ra?

I don't know how jal could cause that problem.  Jal is an assembler alias that just shoves ra into the rd field of the instruction.  It doesn't create any symbols.

MarcKarasek, 2020-03-30:
I see this from nm
                 U ra
Have no idea where it is coming from if this is the problem.. 

I am using ra in a few other places to return from a function call 
jal foo

foo:
..
..
..
j ra

MarcKarasek, 2020-03-30:
The jra seems to be the problem it should be jr ra

MarcKarasek, 2020-03-30:
That fixed it... 
Thanks for the pointer

closed 2020-03-30 by MarcKarasek