riscv-collab/riscv-gnu-toolchain #614: There is an error when I make riscv-gnu-toolchain.

ShiXiongC, 2020-03-30:
I have executed the following command, and try my best to figure out ,now  I'm desperate for help.

cd riscv-tools
cd riscv-gnu-toolchain
sudo ../configure --prefix=/opt/riscv32 --with-arch=rv32imc --with-abi=ilp32d

checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /bin/grep
checking for fgrep... /bin/grep -F
checking for grep that handles long lines and -e... (cached) /bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed

sudo make

rm -rf stamps/build-binutils-newlib build-binutils-newlib
mkdir build-binutils-newlib
cd build-binutils-newlib && CC_FOR_TARGET=riscv32-unknown-elf-gcc /home/csx/riscv-tools/riscv-gnu-toolchain/riscv-binutils/configure \
	--target=riscv32-unknown-elf \
	 \
	--prefix=/opt/riscv32 \
	 \
	--disable-werror \
	--with-expat=yes  \
	--disable-gdb \
	--disable-sim \
	--disable-libdecnumber \
	--disable-readline
checking build system type... i686-pc-linux-gnu
checking host system type... i686-pc-linux-gnu
checking target system type... riscv32-unknown-elf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /bin/sed
checking for gawk... /usr/bin/gawk
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether g++ accepts -static-libstdc++ -static-libgcc... yes
checking for gnatbind... no
checking for gnatmake... no
checking whether compiler driver understands Ada... no
checking how to compare bootstrapped objects... cmp --ignore-initial=16 $$f1 $$f2
checking for objdir... .libs
checking whether to use debuginfod... auto
checking for debuginfod_begin in -ldebuginfod... no
checking whether debuginfod_begin is declared... no
configure: WARNING: debuginfod is missing or unusable; some features may be unavailable.
checking for isl 0.15 or later... no
required isl version is 0.15 or later
*** This configuration is not supported in the following subdirectories:
     readline libdecnumber sim gdb
    (Any other directories should still work fine.)
checking for default BUILD_CONFIG... 
checking for --enable-vtable-verify... no
checking for bison... bison -y
checking for bison... bison
checking for gm4... no
checking for gnum4... no
checking for m4... m4
checking for flex... flex
checking for flex... flex
checking for makeinfo... makeinfo
checking for expect... no
checking for runtest... no
checking for ar... ar
checking for as... as
checking for dlltool... no
checking for ld... ld
checking for lipo... no
checking for nm... nm
checking for ranlib... ranlib
checking for strip... strip
checking for windres... no
checking for windmc... no
checking for objcopy... objcopy
checking for objdump... objdump
checking for readelf... readelf
checking for cc... (cached) riscv32-unknown-elf-gcc
checking for gcc... (cached) riscv32-unknown-elf-gcc
checking for riscv32-unknown-elf-c++... no
checking for riscv32-unknown-elf-g++... no
checking for riscv32-unknown-elf-cxx... no
checking for riscv32-unknown-elf-gxx... no
checking for riscv32-unknown-elf-gcc... no
checking for riscv32-unknown-elf-gfortran... no
checking for riscv32-unknown-elf-gccgo... no
checking for riscv32-unknown-elf-ar... no
checking for riscv32-unknown-elf-as... no
checking for riscv32-unknown-elf-dlltool... no
checking for riscv32-unknown-elf-ld... no
checking for riscv32-unknown-elf-lipo... no
checking for riscv32-unknown-elf-nm... no
checking for riscv32-unknown-elf-objcopy... no
checking for riscv32-unknown-elf-objdump... no
checking for riscv32-unknown-elf-ranlib... no
checking for riscv32-unknown-elf-readelf... no
checking for riscv32-unknown-elf-strip... no
checking for riscv32-unknown-elf-windres... no
checking for riscv32-unknown-elf-windmc... no
checking where to find the target ar... just compiled
checking where to find the target as... just compiled
checking where to find the target cc... pre-installed
checking where to find the target c++... pre-installed
checking where to find the target c++ for libstdc++... pre-installed
checking where to find the target dlltool... just compiled
checking where to find the target gcc... pre-installed
checking where to find the target gfortran... pre-installed
checking where to find the target gccgo... pre-installed
checking where to find the target ld... just compiled
checking where to find the target lipo... pre-installed
checking where to find the target nm... just compiled
checking where to find the target objcopy... just compiled
checking where to find the target objdump... just compiled
checking where to find the target ranlib... just compiled
checking where to find the target readelf... just compiled
checking where to find the target strip... just compiled
checking where to find the target windres... just compiled
checking where to find the target windmc... just compiled
checking whether to enable maintainer-specific portions of Makefiles... no
configure: creating ./config.status
config.status: creating Makefile
make -C build-binutils-newlib
make[1]: Entering directory '/home/csx/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
Makefile:527: *** missing separator.  Stop.
make[1]: Leaving directory '/home/csx/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
Makefile:388: recipe for target 'stamps/build-binutils-newlib' failed
make: *** [stamps/build-binutils-newlib] Error 2

jim-wilson, 2020-03-30:
> sudo ../configure --prefix=/opt/riscv32 --with-arch=rv32imc --with-abi=ilp32d

ilp32d passes double FP values in the double FP registers, but rv32imc has no FP registers at all.  We should probably try to detect this error at configure time.  But this doesn't look like the cause of your build failure.

> make -C build-binutils-newlib
> make[1]: Entering directory '/home/csx/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
> Makefile:527: *** missing separator. Stop.

So what is on line 527 of the Makefile in build-binutils-newlib?

ShiXiongC, 2020-03-30:
> > 须藤../configure --prefix = / opt / riscv32 --with-arch = rv32imc --with-abi = ilp32d
> 
> ilp32d在双FP寄存器中传递双FP值，但是rv32imc根本没有FP寄存器。我们可能应该尝试在配置时检测到此错误。但是，这似乎不是导致构建失败的原因。
> 
> > make -C build-binutils-newlib 
> > make [1]：输入目录'/ home / csx / riscv-tools / riscv-gnu-toolchain / build-binutils-newlib'Makefile 
> > ：527：***缺少分隔符。停止。
> 
> 那么build-binutils-newlib中Makefile的第527行是什么？

[image]

I want to install riscv compiler to generate.RISCV64 suffix files，however, the RISCV environment has not been successfully installed

ShiXiongC, 2020-03-30:
> ilp32d在双FP寄存器中传递双FP值，但是rv32imc根本没有FP寄存器。我们可能应该尝试在配置时检测到此错误。但是，这似乎不是导致构建失败的原因。
>
>
> 那么build-binutils-newlib中Makefile的第527行是什么？
526 #### host and target specific makefile fragments come in here.
527 @target_makefile_frag@
528 @alphaieee_frag@
529 @ospace_frag@
530 @host_makefile_frag@

jim-wilson, 2020-03-30:
527 @target_makefile_frag@

That suggests that sed failed which is odd as it is a pretty simple program.  This could fail if you are using a non-GNU sed, but the config.log says you are building on linux so that seems unlikely.  Still what does "sed --version" say?  I suppose another option from the chinese text is that there could be a locale problem.  Maybe try doing something like "LANG=C.UTF-8 configure" or "LANG=en_US.UTF-8" configure" to see if that fixes the Makefile problem?

By the way, you don't need sudo for the configure, and you only need sudo for the make if installing into a system directory.  it is much safer to do the build without sudo, installing into a dir you have write access to, and then use sudo to copy to a system directory, or else give yourself write permission to the install dir before the build.

cheerryio, 2020-09-17:
Encountered same issue. I clone this repository under the *Windows* and triy to copy the cloned repository to Linux to do the make job and other work . However,due to Git's default configuration ```core.autocrlf true```, when  using ```git clone``` on WIndows, line break types of the cloned files would be CRLF which Windows uses instead of LF which unix and linux use. The first time I run the command ```./configure --prefix=/opt/riscv```,it shows something is wrong with the file ```./configure```.I use vscode to change line break types of several files from CRLF to LF by hand.But more problems still exist in the other files. To solve the problem , run the following commands under repository's directory,which would replace CRLF to LF in all files. 
```
sudo apt install dos2unix
find . -type f -print0 | xargs -0 dos2unix -f
```
Then continue with:
```
./configure --prefix=/opt/riscv
make linux
```
Hope it works.

TommyMurphyTM1234, 2022-09-27:
No update/activity in two years and no specific issues with the RISC-V GCC toolchain or build process itself identified.

closed 2022-09-27 by TommyMurphyTM1234

gxflying, 2023-10-13:
it works,  I have ecounter the same issus, and fix it by using dos2unix tool, 
> Encountered same issue. I clone this repository under the _Windows_ and triy to copy the cloned repository to Linux to do the make job and other work . However,due to Git's default configuration [...]

TommyMurphyTM1234, 2023-10-13:
It might be better in this situation to use something like described here:

* https://docs.github.com/en/get-started/getting-started-with-git/configuring-git-to-handle-line-endings?platform=windows

Or simply don't check out on one platform (e.g Windows) if using on another (Linux, Mac) where the line endings are different.